bbcmicrobit/micropython #388: clear() method for Image objects

jbenc, 2016-12-21:
When using an Image object as an off-screen buffer (see #387), it would be useful to recycle the same buffer instead of creating a new object many times a second (granted, the garbage collector seems to do a good job but still...). A clear() method that sets all pixels to 0 (or, perhaps, to a specified value) would come very handy for this.

dpgeorge, 2016-12-22:
The idea is that you would create a new Image(5,5) for each frame.  But considering that images are in fact mutable and have the set_pixel method, I'd say a fill method would be a very sensible addition.  Eg:
```
img.fill(0) # clear the entire image
img.fill(5) # set the image half-bright
```

dpgeorge, 2017-01-10:
The fill() method is implemented by #390.

closed 2017-01-10 by dpgeorge