bbcmicrobit/micropython #389: Micro:bit and Bluetooth

githubhero, 2016-12-28:
I red the following [here](http://microbit-micropython.readthedocs.io/en/latest/ble.html?highlight=bluetooth):

> While the BBC micro:bit has hardware capable of allowing the device to work as a Bluetooth Low Energy (BLE) device, it only has 16k of RAM. The BLE stack alone takes up 12k RAM which means there’s not enough room to run MicroPython.
> 
> Future versions of the device may come with 32k RAM which would be sufficient. However, until such time it’s highly unlikely MicroPython will support BLE.

I'm a little confused since I don't understand how this thing could be coded without bluetooth. Did he use another technology (NOT micropython)?

https://www.youtube.com/watch?v=aep_GVowKfs

Neon22, 2016-12-28:
suggest you ask this kind of question on the forum

githubhero, 2016-12-28:
Hi @Neon22 , thanks for your kind reply. 

I'm a microbit newbie and I don't know of any forum: could you please point me to its URL?

ZanderBrown, 2016-12-28:
Bluetooth is NOT required to program the Micro:Bit. That is done over USB connection the same way as the online editors.

githubhero, 2016-12-28:
@ZanderBrown USB? I see no cable: how the device and the microbit communicate?

Neon22, 2016-12-28:
The forum is here:
https://forum.micropython.org/viewtopic.php?t=1042

The microbit has a mini usb connector on it. plug it in. it wil be powered by this port and communicate over it.

Neon22, 2016-12-28:
Here's some useful links:
https://www.python.org/community/microbit/
and
https://microbit-micropython.readthedocs.io/en/latest/
https://www.microbit.co.uk/python-guide
https://www.microbit.co.uk/getting-started/python

githubhero, 2016-12-28:
Maybe [this](https://mail.python.org/mailman/listinfo/microbit) is the right place to ask, thanks

githubhero, 2016-12-28:
@Neon22 Thanks for the links. I don't understand the USB-related comment: the connection in that video is apparently done with BLE (are you suggesting the author is using a USB BLE dongle?)

tjguk, 2016-12-28:
Since there seems to be some confusion... the youtube video the OP is 
pointing to is clearly labelled "basic bluetooth capabilities demo" and 
is using an Android app to talk to an otherwise unconnected Micro:bit. 
Presumably via bluetooth.

So, to answer the original question, this is presumably *not* done via 
MicroPython. But MicroPython is not the only toolkit available to talk 
to Micro:bits.

githubhero, 2016-12-28:
@tjguk This is exactly the reply I was searching for, thanks!

jezdean, 2016-12-28:
@githubhero: I'd perhaps not ask on the mailing list. Although they're very friendly, it tends to be about mechanics of micropython on the microbit.

The Microbit Blue app is in the app store. The microbit runs a `.hex` file which I believe is written in pxt.io. The [source code to the app is here](https://github.com/microbit-foundation/microbit-blue). The `.hex` files are there also.

To get started with Bluetooth and the microbit look to use evothings. [Here's an example of it using the microbit.](https://evothings.com/doc/examples/microbit-sensors.html). Here's a [write up of using the microbit with evothings](https://evothings.com/evothings-and-the-bbc-microbit/).

Eventually you might want to look at Apache Cordova.

githubhero, 2016-12-28:
@jezdean When I found that video, my idea was to connect iOS and the micro:bit to do the same things you see in the video using Swift 3 and BLE. 

Since my first Micropython experiments went really fine, I thought Micropython was the ok: then, I found the lines I posted.

I must admit I initially thought that any technology would let you do the same things: I didn't suspect, for example, pxt.io (which apparently is in beta) could do things micropython could not do (and viceversa).

Thanks for the precious tips.   :-)

githubhero, 2016-12-28:
@jezdean It's not PXT, it's something else (still don't know what): https://github.com/microbit-foundation/microbit-blue/issues/1#issuecomment-269469986   :-)

markshannon, 2016-12-28:
Indeed.

Everyone, the issue tracker is not a forum.
Please use one of the email lists for further discussions.

closed 2016-12-28 by markshannon

githubhero, 2016-12-28:
@ZanderBrown @markshannon I don't know if we could define "not an issue" the lack of Bluetooth, but anyways, no problem, I posted here because I could not find the forums and I didn't know where to ask my question

githubhero, 2016-12-28:
@whaleygeek Not sure: the programs at developer.mbed.org includes "mbed.h" but I see no "MicroBit.h". 

[Here](http://www.bittysoftware.com/tutorials/bitty_audio_prank_yotta.html)'s what he probably uses:

> To program your micro:bit using C/C++ you need the Yotta tools and the micro:bit 'DAL' source code on a PC or Mac. If you haven't yet set this environment up, read the information on our tools page and make sure you can build and install the Hello World sample code onto your micro:bit before going any further. If you've accomplished this then you have a working C/C++ environment for micro:bit development and can move on.