bbcmicrobit/micropython #391: PWM Gets Corrupted using Mu

4tronix, 2016-12-30:
I have been writing various test scripts for the 4ronix Bit:Bot and have noticed a serious problem with PWM which is used to control the speed of the motors.

First spotted when writing a line following program that flashed different colours of neopixels depending what was happening with the line. Changing the neopixels randomly caused the motors to stop – the PWM clock seems to have stopped running.
Simply by commenting out the np.show() line and the program worked as expected. Similarly, removing the PWM from the motors allowed the motors to keep going even with the line np.show() enabled.

I also noticed an identical problem (PWM for motors stopped running) when using the running_time() method to determine the distance to an object with an HC-SR04. Again this randomly stopped the motors. Removing the PWM from the motors (ie. making them fully on or fully off) allowed the program to work.

Someone mentioned a different PWM issue when using Mu, so I tried the *exact same* program using the online micropython editor. All works absolutely fine.

Try running the program at www.4tronix.co.uk/bitbot/sonic02.py using Mu and then using the online editor. Or the program www.4tronix.co.uk/bitbot/LineFollower2.py . Both work fine when using the online editor and both randomly stop the motors when using Mu.

I believe that the 4-line complete example below is related. With Mu, the motor keeps turning. With the online editor, it stops after the 2 second sleep.
`from microbit import *
pin8.write_digital(0)
pin0.write_analog(100)
sleep(2000)
pin0.write_digital(0)
`

ZanderBrown, 2016-12-31:
What we are saying here is that between the build used by the online editor are the build included in uFlash PWM has gone haywire

whaleygeek, 2016-12-31:
So, the first step is to identify the githash of the source for the version
inside the online editor, and the githash of the source for the version
embedded inside uFlash.

We then have a chance to look through the git commit logs to see what
change caused the error, as all the source is meticulously controlled in a
version control system.

jezdean, 2016-12-31:
@whaleygeek The runtime in uFlash was last updated [here](https://github.com/ntoll/uflash/commit/fe26fe8ea0b7943674cb8165e6939cba3ef0dc88).

I think it lacks [this update of micropython for the microbit](https://github.com/bbcmicrobit/micropython/pull/345).

I submitted the issue to the uFlash  repo.

DavidWhaleIET, 2017-01-11:
Damien has suggested (somewhere else, on the mailing list I think?) to build with the new firmware and try that. But building new firmware is a bit too advanced for our user in this case, as is using the hexlify tool to merge the user script with it, as they do not have python installed. If there is someone else that can 'try this with the latest firmware and get back to us' that would be really helpful, thanks.

ntoll, 2017-01-11:
This is in hand. But we need to fix https://github.com/bbcmicrobit/micropython/pull/397 before releasing a new version of the runtime.

DavidWhaleIET, 2017-01-11:
Ok thanks. See related: https://github.com/mu-editor/mu/issues/227

4tronix, 2017-01-11:
Thanks for the updates. I look forward to testing the new release

4tronix, 2017-02-01:
Is there any update to this issue?

ntoll, 2017-02-01:
A new release will be coming out soon. We've had to squash a number of bugs (still ongoing) before the runtime is ready. I'm testing things this week.

DavidWhaleMEF, 2017-02-13:
i @ntoll any ETA on surfacing this fix yet? Thanks.

ntoll, 2017-02-13:
You'll be the first to know.

DavidWhaleMEF, 2017-03-06:
Hello, I'm getting new reports of PWM problems in Mu, and I'm not sure what to tell people.

The report today is that PWM worked fine in the old BBC python editor, but that it's now broken in python.microbit.org and codewithmu.com

As we have removed the links to the old BBC python editor from the microbit.org web site this week, this is now forcing this issue to be high priority as people can't do motor speed control properly with any of the actively deployed python editors.

microbit.org-ticket: 1245

dpgeorge, 2017-03-07:
> The report today is that PWM worked fine in the old BBC python editor, but that it's now broken in python.microbit.org and codewithmu.com

Please can you provide a copy of the .hex file that has the problem, along with the script that does PWM, and a description of the symptoms.

ntoll, 2017-03-07:
We're aware of this problem, I believe Mark has already fixed it and it's merged into master. Just saying "it's broken" isn't helpful. As @dpgeorge mentioned, we need the script that does the PWM and a description of what's going on (what did you expect to happen? what actually happened? why is this a bug?) for us to make any useful progress on this and prove the PWM fix actually works.

In terms of making a release, there are various bits and bobs that need to line up in terms of bug fixes, reviews and people finding time to work on this.

DavidWhaleMEF, 2017-03-07:
I have tried to encourage the users that reported this issue to report it directly in github here, but they have not followed my advice. For data protection reasons, I cannot share user scripts without their permission. There is only so much I can do, sorry.

DavidWhaleMEF, 2017-03-07:
Okay, so you say it's 'fixed in master', My wider question then is 'has master been deployed into the latest Mu and python.microbit.org' yet so that users can check that the fix works for them, or is there a patch they can download to confirm that the fix works for them and they can then report back to you? Thanks.

DavidWhaleMEF, 2017-03-07:
I see also that @4tronix has provided .py scripts as above on Dec 30th - have you been able to test with those and did they work or do they still fail with the new fix? Thanks

deshipu, 2017-03-07:
Hi @DavidWhaleMEF,

it takes a lot of work to make a release, and you can't just make it at arbitrary points of time during development. You have to make sure everything works correctly and all the changes work well together, otherwise you would get a perpetually broken product. It doesn't help that there are only a few people who are trusted to make the release, and unsurprisingly those are the same people that have a lot of other things on their plate. So please be patient.

On the other hand, I don't think it's really such a great effort to write a few lines of Python code that would demonstrate the issue and allow testing by the developers (instead of doing the whole release process just so that one person can re-run their script when they feel like it). You don't have to share their programs (in fact, you shouldn't) -- instead, isolate the issue and write a short example that fails. That would be really helpful for everyone, and would make it possible to test the fix before doing all the work for the release, making sure it is really fixed.

We really appreciate the time you devote to reporting the issue and to requesting status updates -- it shows you really care. Perhaps you could lend us a hand and help get this fixed faster.

DavidWhaleMEF, 2017-03-07:
Sure, thanks. But the user script is at the top of this issue, has anyone been able to test that with the current 'master' yet and confirm if it is fixed or broken still?

Thanks.

microbit-mark, 2017-12-13:
Hey @4tronix just refreshing this issue as we're going through the backlog. Have you been able to test the scripts on Bit:Bot with the current master branch?

DavidWhaleMEF, 2017-12-13:
Do we have any instructions anywhere on how a user would use the latest master branch? Using master requires it to be built from source, which is non trivial for most users.

The version1 branch (which is not master) is being managed by @dpgeorge and I think he has a cached .hex somewhere that can be used. We also need instructions for users on how to use a custom .hex and use that alongside their script (as python.microbit.org and codewith.mu will both use the embedded micropython .hex file and not the custom one). So we need a k-base I think on how to do this, so that we can enable our users to test with these new draft releases. @carlosperate may be able to help with those instructions.

4tronix, 2017-12-13:
Thanks David,
I was wondering how to say “I don’t know how” !

Gareth

From: David Whale (Micro:bit Educational Foundation)
Sent: 13 December 2017 12:28
To: bbcmicrobit/micropython
Cc: 4tronix; Mention
Subject: Re: [bbcmicrobit/micropython] PWM Gets Corrupted using Mu (#391)

Do we have any instructions anywhere on how a user would use the latest master branch? Using master requires it to be built from source, which is non trivial for most users.
The version1 branch (which is not master) is being managed by @dpgeorge and I think he has a cached .hex somewhere that can be used. We also need instructions for users on how to use a custom .hex and use that alongside their script (as python.microbit.org and codewith.mu will both use the embedded micropython .hex file and not the custom one). So we need a k-base I think on how to do this, so that we can enable our users to test with these new draft releases. @carlosperate may be able to help with those instructions.

DavidWhaleMEF, 2017-12-13:
The other way we have done it before is to use @ntoll PythonEditor: https://github.com/bbcmicrobit/PythonEditor

If you can follow the install instructions in the readme (you need Python on your machine) and then replace the firmware.hex file with the draft replacement .hex, you can then paste your code into the local web browser editor that this serves up for you, and it will generate a .hex with the draft MicroPython and your script combined that will run on the micro:bit

https://github.com/bbcmicrobit/PythonEditor

Alternatively, @carlosperate knows a way to use the uflash tool from the command line to do the same, if you are comfortable working at the command line.

DavidWhaleMEF, 2017-12-13:
@microbit-mark The draft version1 hex from @dpgeorge can be found here:

https://github.com/bbcmicrobit/micropython/issues/472#issuecomment-334364423

Would you have time to work through all this and write a k-base for us so that users can follow the instructions when needing to re-test with a newer draft release?

carlosperate, 2017-12-13:
> Do we have any instructions anywhere on how a user would use the latest master branch? Using master requires it to be built from source, which is non trivial for most users.

The current beta version of the online editor (http://python.microbit.org/v/beta) has the v1.0 branch at e17de0954178d88aa2a1d70b6e6ebc43f5456607 (latest commit at the time of writing).

microbit-mark, 2017-12-13:
@whaleygeek Yep I can do that for testing. We already have this https://support.microbit.org/solution/articles/19000044801-how-to-use-an-updated-micropython-image for the online editor and the 'beta', but how might we test a specific branch with Mu?

carlosperate, 2017-12-13:
@microbit-mark @DavidWhaleIET For how to do it in Mu we should continue that conversation in the mu repository: https://github.com/mu-editor/mu/issues/313

carlosperate, 2017-12-13:
@4tronix in the meantime, would you be able to test in http://python.microbit.org/v/beta the code that showed this issues?

microbit-carlos, 2018-03-02:
@4tronix just wondering if you had a chance to test this one with http://python.microbit.org/v/beta ?

dpgeorge, 2018-03-07:
This should be fixed since merging version1 in e26d7c89d4a96de0fa0a1dd5aec024b31fc4816e

closed 2018-03-07 by dpgeorge