bbcmicrobit/micropython #393: What is the difference between microbit-micropython.hex and frimware.hex?

KimHwaMok, 2017-01-01:
What is the difference between microbit-micropython.hex and frimware.hex?

deshipu, 2017-01-01:
Please ask such questions on the forum at http://forum.micropython.org, this is for reporting problems with the code.

KimHwaMok, 2017-01-01:
hi @deshipu ! :)
My problem is here!
I want to change the microbitdisplay.cpp module.

This is a module associated with microbit.display.scroll.
--------------------------------------------------------------------------
DEFAULT_SCROLL_SPEED 150

void microbit_display_scroll(microbit_display_obj_t *self, const char* str) {
mp_obj_t iterable = scrolling_string_image_iterable(str, strlen(str), NULL, false, false);
microbit_display_animate(self, iterable, DEFAULT_SCROLL_SPEED, false, true);
}
--------------------------------------------------------------------------
and i made example simple code
import microbit
microbit.display.scroll('hello!)
dir(microbit)
--------------------------------------------------------------------------
and i changed micropython display module
DEFAULT_SCROLL_SPEED 150

void microbit_display_scroll(microbit_display_obj_t *self, const char* str) {
const char *tmp = "test version!"
mp_obj_t iterable = scrolling_string_image_iterable(tmp, strlen(tmp), NULL, false, false);
microbit_display_animate(self, iterable, DEFAULT_SCROLL_SPEED, false, true);
}
--------------------------------------------------------------------------
and rebuild!
The output must be "test version!", but the output is still "hello!"
what is wrong in my code?

closed 2017-01-02 by KimHwaMok