bbcmicrobit/micropython #398: ValueError: pin0 is in music mode

shenki, 2017-01-11:
With code that looks like the following, we have occasionally seen `ValueError: pin0 is in music mode`:

```python
while True:
  music.play(music.NYAN, wait=False)
```

dpgeorge, 2017-01-11:
@markshannon I took a quick look at the code and it seems like microbit_obj_pin_acquire will raise an exception when trying to set a pin to the mode that it's already in.  I'd say it should succeed in this case.

Also, these pin acquire/free functions should be surrounded by IRQ disable/enable pairs if they are going to be called from the ticker IRQ.

tsukasa-au, 2017-01-11:
I have found that by forcing the async music state back to idle, then calling microbit_obj_pin_release before setting up the async state again.

You can see this in 704ba76320b3be8a41798a253d50acffed30efe9.

dpgeorge, 2017-01-15:
This is still a problem with music.pitch, eg:
```
>>> import music
>>> music.pitch(440)
>>> music.pitch(440)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin 0 in music mode
```

PR #399 (2017-01-15): Fix ValueError exception that occurs when playing a second song.
We now explicitly stop the current song that is playing before
attempting to start a new song.

Fixes #398
PR #405 (2017-02-10): Sort out pin modes and music module
Fixes #398

This PR:

1. Changes pin-mode implementation to use pointers to structs, rather than strings. This should avoid any race conditions when releasing a pin in an interrupt handler.
2. Documents which pin-mode functions can and cannot be called in an interrupt handler.
3. Improves the pin-mode API doing the sensible thing when passed a NULL or attempting to re-acquire the same pin.
4. Fix music module's pin releasing.
5. Adds a test for the music module (just tests that it doesn't crash, not that it makes the right noises :musical_note: ).

Includes  [... cut]
closed 2017-02-10 by dpgeorge