bbcmicrobit/micropython #400: Can extra functions be exposed

mrpotes, 2017-01-15:
It's been a long time since I used C/C++, and I'm pretty new to Python, but I notice that [`inc/microbit/microbitimage.h`](https://github.com/bbcmicrobit/micropython/blob/master/inc/microbit/microbitimage.h#L69) has a function declaration, `microbit_image_for_char`, that sounds like just what I need - could it be exposed as a python function for turning a single character into the LED image string, or maybe a string into an array of LED image strings?

Thanks :)

ZanderBrown, 2017-01-15:
is `SHOW 'a'` not sufficient?

deshipu, 2017-01-15:
You can do better:

```
from microbit import Image

letter_a = Image('A')
```

mrpotes, 2017-01-15:
No, we need to modify the LED arrangement so that it will read correctly in a mirror, but start from a normal letter representation to do so, so it would be great not to have to reproduce something the microbit already knows how to do.

deshipu, 2017-01-15:
Why do you need a string for it? Can't you invert the image?

mrpotes, 2017-01-15:
@deshipu does `Image('A')` work? The docs say:
> If string is used, it has to consist of digits 0-9 arranged into lines, describing the image, for example:

deshipu, 2017-01-15:
If there was only some way of checking if it works...

mrpotes, 2017-01-15:
@deshipu we want the text to scroll just like it does if you use the scroll function, but mirrored. As I understand it, `invert` inverts the brightness of the image (0 -> 9, 1 -> 8, etc), it doesn't do flip horizontally (or vertically, for that matter)

mrpotes, 2017-01-15:
> If there was only some way of checking if it works...

If only the microbits weren't locked away in a cupboard at school, you mean?

deshipu, 2017-01-15:
Yes, you will have to write the function that mirrors the image yourself. But it should be easier and faster with an image than with a string.

deshipu, 2017-01-15:
> If only the microbits weren't locked away in a cupboard at school, you mean?

I'm sorry, I didn't think.

mrpotes, 2017-01-15:
> I'm sorry, I didn't think.

No problem :)

So if Image('A') works that's great, then I'm thinking something like (pseudo code as I haven't worked out all of this in python):

```
def scrollMirrored(str):
  show(
    chars(str)
      .map(c: repr(Image(c)).split(':'))
      .zip()
      .map(str_join)
      .map(s: s.reverse)
      .map(s: chars(str).slidingWindow(5))
      .zip()
      .map(arr: arr.join(':')))
```