bbcmicrobit/micropython #404: New function radio.is_on()

hickford, 2017-01-19:
Hi. While the display module has functions `on()` `off()` and [`is_on()`](https://microbit-micropython.readthedocs.io/en/latest/display.html#microbit.display.is_on), the radio module has only `on()` and `off()`. Have you considered adding an `is_on()` function to query the state? 

My program turns the radio off in certain modes. Currently I keep track of that in a separate variable.

ntoll, 2017-01-19:
Hmm... I'm playing devil's advocate here - perhaps its a good thing to keep track of mode in the separate variable rather than checking if the radio if on? Or am I missing how your code works? Perhaps you could share your use case?

In not against adding an `is_on()` function to the radio module, but I'd like to understand your use case further.

hickford, 2017-01-19:
Hi Nicholas. Thanks for your reply. My two microbits function as a remote control and receiver. They remote only sends, the receiver only receives. Nevertheless, so I can't mix them up, they run the same code. Check for button press, send message, check for received message, act on it, repeat.

    if button_b.was_pressed():
        radio.send("B")

    received = radio.receive()
    if received:
        act()

This worked fine. Then I had an idea to save power on the remote, which runs on battery. The idea was to turn off the radio between button presses. It's necessary then to skip `radio.receive()` when the radio is off, or it raises an error. Hence I'd like to test if the radio is on

    if radio.is_on():
        received = radio.receive()

It was easy to workaround with a separate variable.