bbcmicrobit/micropython #409: What encoding method do micropython use?

KimHwaMok, 2017-02-10:
My development environment is Ubuntu 16.04 [64bit].
I am developing micro-bit that supports multiple languages.
What encoding method do micropython use in micro:bit?

dpgeorge, 2017-02-10:
Unicode strings are encoded using utf8.

KimHwaMok, 2017-02-10:
Hi, @dpgeorge Thank you for reply!
Can i use [setlocale, mbtowc] function in mycropython?

dpgeorge, 2017-02-10:
> Can i use [setlocale, mbtowc] function in mycropython?

No, there is no concept of locale.  You'll need to implement it yourself in (Micro)Python.

KimHwaMok, 2017-02-13:
@dpgeorge Thank you for reply again!
In source/microbit/microbitdisplay function, I implemented it like this:

`#include <locale.h>
setlocale(LC_ALL, "");
mbtowc(0, 0, 0);`
The locale function is defined in locale.h. 
And when I rebuild using "make all", I did not get an error.
Do I still have to implement the locale function myself?

dpgeorge, 2017-02-13:
I don't know if setlocale or mbtowc is implemented by the provided stdlib (newlib).  If you are going this deep then you need to try it out for yourself and see what works.

KimHwaMok, 2017-02-13:
@dpgeorge thank you again.
Okay. I got it. After the test, we will inform you of the result.

KimHwaMok, 2017-02-16:
Hi @dpgeorge !
I think "setlocale" function in "<locale.h>" is defined in MicroPython but not implemented yet.
so, I have to implement that function in MicroPython. (setlocale)

Do you have any idea about implement setlocale function in MicroPython?

dpgeorge, 2017-02-16:
> Do you have any idea about implement setlocale function in MicroPython?

No, sorry.

closed 2017-04-07 by dpgeorge