bbcmicrobit/micropython #413: Temperature calibration, as supported in the runtime

SolarPolarMan, 2017-02-28:
Could we see support for temperature calibration [since it's in the runtime](https://lancaster-university.github.io/microbit-docs/ubit/thermometer/#int-setcalibration-int-calibrationtemp)? If not, could you shed some light on how the temperature calibration function works so I can imitate it?

dpgeorge, 2017-02-28:
> If not, could you shed some light on how the temperature calibration function works so I can imitate it?

I think it's simply taking the difference between the measured and actual value, then adding that to subsequent measurements.  Eg:
```python
def calibrate_temp(t):
    offset = t - temperature()
def get_calibrated_temp():
    return temperature() + offset
```

SolarPolarMan, 2017-02-28:
So we just assume that the chip stays at the same temperature, relative to the environment?

dpgeorge, 2017-03-01:
> So we just assume that the chip stays at the same temperature, relative to the environment?

No, the chip changes temperature as the environment changes.  It should be quite accurate in terms of relative temperature change, but not the absolute value.  Hence given 1 data point (the calibration value) you should be able to compute an accurate absolute temperature.