bbcmicrobit/micropython #414: Audio module docs does not appear in contents page

DavidWhaleMEF, 2017-03-08:
The audio module does not appear in the contents page of the docs. If I search for Audio, it finds the docs page, but if I look at the scrolling list on the far left of the docs, it is not there.

microbit-rosslowe, 2018-10-05:
`checking consistency... /.../micropython/docs/audio.rst: WARNING: document isn't included in any toctree`

Adding to the 'API reference' section in index.rst should resolve this:
[image]

See PR above.

microbit-carlos, 2018-10-16:
This is now covered in PR https://github.com/bbcmicrobit/micropython/pull/578 instead of #573.

DavidWhaleMEF, 2018-12-13:
Do we have an ETA of when the readthedocs pages will go live with this fix @carlosperate .. Thanks :-)

closed 2018-12-13 by microbit-carlos

DavidWhaleMEF, 2018-12-13:
Excellent work, well done team!!!