bbcmicrobit/micropython #415: audio module documentation does not match the code (pins parameter)

DavidWhaleMEF, 2017-03-08:
The documentation for Audio says it has a 'pins' parameter, taking a tuple of two pins to connect the speaker to...
[image]

However, if I use the pins parameter like described in the docs, I get an error.

>>> audio.play(repeated_frame(triangle, 1500), pins=(pin0, pin1))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: extra keyword arguments given

If I however use a pin= parameter with a single pin, it does indeed play the AudioFrame on that one pin.

audio.play(repeated_frame(triangle, 10), pin=pin1)

Is it no longer possible to play audio frames on two pins?

DavidWhaleMEF, 2017-03-08:
Note, if I press the HELP button in Mu, it tells me this is the version I have installed...

https://codewith.mu/help/0.9.12/

I can't see anything on the codewith.mu web page that tells me what the latest version number is, to know whether I have the latest version installed or not.

ZanderBrown, 2017-03-08:
Working out Mu version is something we have been discussing.

By checking the releases on Mu's GitHub page you can see the latest is `v0.9.13`

DavidWhaleMEF, 2017-03-08:
Yeah, that's why I pressed the help button, it was you or carlos that suggested that (useful!) temporary trick, thanks!

We can't expect teachers and 11 year old children to instinctively know where the github pages are or indeed how to interpret them. I think if we have a downloads page, it should be obvious what the version number of the download is, so that you can check whether the version you have installed needs updating or not. Especially if we regularly use the line 'are you using the latest version' - it needs to be obvious for the target audience how they check if they have the latest version or not.

Otherwise you have to download the package and run it to work out what version the latest one is, and then press the HELP button in your old Mu and see what web page it goes to to see if you have the latest, and press the HELP button in the new one to see what version it is.

I'm sure it would be possible to make this (common) workflow much simpler for the users, don't you think? Thanks.

ZanderBrown, 2017-03-08:
[image]
@DavidWhaleMEF off topic but are you seeing some odd timestamps on @GitHub

DavidWhaleMEF, 2017-03-08:
@ZanderBrown No, I think it's your end only, I see this...

[image]

ZanderBrown, 2017-03-08:
interesting i see this page as:
[image]

DavidWhaleMEF, 2017-03-08:
Hey @ZanderBrown you've just invented the time travel module! Quick, check it into github and get it into the next release!!!

ntoll, 2017-03-08:
Folks, just a polite notice to NOT mix up issues.

Mu issues belong in the Mu repos, MicroPython issues belong here and cute photos about GitHub really being run on a TARDIS should perhaps go on the microbit@python.org mailing list for everyone to enjoy. ;-)

While *I* know it's fun, and I'm often guilty of this myself, given the lack of time many of us currently have, trying to wade through a large number of emails/messages that are orthogonal to the original issue just takes up time.

To be clear, I'm not being a meanie and saying don't report issues or post GitHub screenies -- we want this project to be fun for all concerned! Rather, so that some poor bugger doesn't have to wade through all the messages only to discover they're fun chatter rather than more details of an audio module documentation bug, then can we just think about the medium we use for such friendly chatter!

Thanks!

:-)

DavidWhaleMEF, 2017-03-08:
@ntoll - the original issue is a micropython issue though - the docs don't match the code - this is nothing to do with Mu. I only provided the mu version so that there was *some* chance of someone tracing back to the version of micro:python that I am referring to!

ntoll, 2017-03-08:
:man_shrugging: Sure, but then the conversation veered off onto how you can tell what version of Mu is being used.

microbit-carlos, 2018-03-14:
Looks like the function signature documentation needs to be changed to:
```
audio.play(source, wait=True, pin=pin0, return_pin=pin1)
```

@DavidWhaleMEF could could check if that resolves your original issue?

DavidWhaleMEF, 2018-03-14:
@carlosperate That was a long time ago! I'm happy that if you have tried this and it works, then it's a docs issue. I'll note for future reference if I use this feature again to try it, but the project was so long ago now i forget what I was even trying to do! Thanks.

closed 2018-10-26 by jaustin