bbcmicrobit/micropython #416: music module docs do not match code (music.stop)

DavidWhaleMEF, 2017-03-08:
The docs say that the music.stop() function has a pin=parameter

[image]

However, if I use a pin parameter, I get an error.

>>> music.stop(pin=pin1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function does not take keyword arguments

If I however just put the pin name, it works

music.stop(pin1)

The docs don't match the code.

Apparently using: http://codewith.mu/help/0.9.12

ZanderBrown, 2017-03-08:
I believe the docs is showing that it has a default value

DavidWhaleMEF, 2017-03-08:
Thanks. Although it confused me, and I'm supposed to know what I'm doing :)

DavidWhaleMEF, 2017-03-08:
Remember that this documentation is for a target age group of 11 year old children. Anything we can do to make it more accessible and obvious to them is a good thing :)

ntoll, 2017-03-08:
Hi @DavidWhaleMEF, to be clear, this is *NOT* documentation for 11yo children, it's documentation for developers, adults and teachers who may support 11yo children and other beginner programmers. 

I've always said (to the BBC at least) that we trust these supervising / supporting people will know their students best and know how to most effectively explain Python. We're just giving them to technical know-how to create their own resources. Stuff like the microbit-playground is more geared to resources beginner programmers may actually end up seeing.

Thanks for all the feedback, keep it coming! :-)

DavidWhaleMEF, 2017-03-08:
Okay, but we will need documentation for 11 year old children at some point, as they are the target audience. Add it to the list, and I'll help out with it when I can :)

And to be clear, adults (teachers and volunteers) find the whole (*kwargs) style Python documentation very confusing (they tell me as such), so it's not really written for them either?

ntoll, 2017-03-08:
I hear you!

Here's how things are organised:

* Teachers and volunteers should read the tutorials - they are written for adults who have a basic understanding of computers and if started at the beginning and completed to the end, they'll pick up quite a lot of Python (as well as micro:bit) knowledge too.

* Developers should read the API documentation (which is what this ticket is about). This is for experienced Python developers and assumes familiarity with Python and its idioms. We follow the same documentation style as the main Python docs on python.org.

We will be moving the tutorials out from these docs and into the pythonineducation.org website (under the micro:bit section). This will also include efforts in translating them to other languages. The development of this website is currently stalled while I try to hit my book deadline -- although you'll be able to see the current "work in progress" at the URL I link to above or pyedu.io.

microbit-carlos, 2018-10-19:
Implemented in PR #572, thanks everybody!

closed 2018-10-19 by microbit-carlos