bbcmicrobit/micropython #417: music.pitch only allows one pitch to be played at a time

DavidWhaleMEF, 2017-03-08:
The music.pitch module apparently allows you to play a pitch on multiple pins simultaneously.

music.pitch(440, pin=pin0)
music.pitch(600,pin=pin1)
music.stop(pin0)
music.stop(pin1)

At line 1, the tone 440 is generated on pin 0
However, at line 2, the tone 600 is generated on both pin0 and pin1.
At line 3, the tone on pin0 is stopped, and 600 is still generated on pin1
At line 4, the tone on pin1 stops.

It's not clear to me from the docs, which indicate by the fact that you can pitch and stop any arbitrary pins, whether this is *supposed* to allow generation of different tones on different pins or not. If it is not supposed to support it, could you please add some clarification to the docs saying that MicroPython can only generate one tone at a time, and generating tones on two or more pins will only generate the last requested tone. If however this is a bug and it *should* be able to generate different tones on different pins, can you mark it as a bug and add it to the queue? Thanks.

The micro:bit nRF51 processor supports up to three simultaneous PWM modules, but I understand from talking to the Lancaster guys that the period of these units is shared throughout the system. So I *suspect* it is impossible to play multiple pitches simultaneously, but I think the documentation should be clarified to make this point more obvious - the fact that you can start a pitch playing on different pins simultaneously leads the user to 'expect' that the pitch generated on each pin can be different, and it was a surprise to me when it wasn't possible (which caused me to clarify the PWM architecture with the Lancaster team).

Apparently using this editor: https://codewith.mu/help/0.9.12/

markshannon, 2017-04-18:
To clarify, the nRF51 processor does not support PWM directly at all. At the hardware level it is the timers that drive PWM.

For a reasonably efficient implementation each period requires one timer. PWM can then be built on the timer using interrupts or linking the clock directly to the pin using the GPIOTE and PPI components. We use the former, the DAL the latter; it makes little difference in terms of efficiency and we need the GPIOTE and PPI components for the audio module.

DavidWhaleMEF, 2017-04-19:
Sure. My point was that I think the docs are a bit unclear, and could do with perhaps a note to say that the current implementation allows only one pitch to be played on one pin at any one time. The API leads the user to think that you could generate two independent pitches on two or more pins.

PR #574 (2018-10-19): docs: update music.rst to include clarification on music:pitch
Add clarification to music:pitch in docs, closes #417 

"Note that you can only play one pitch on one pin at any one time."
closed 2018-10-19 by microbit-carlos