bbcmicrobit/micropython #418: Speech does not work after using music.pitch

DavidWhaleMEF, 2017-03-08:
mu version: 0.9.13

import music
import speech

speech.say("lift")
sleep(1000)
music.pitch(440,pin=pin1)
music.stop(pin1)
speech.say("lift")

The first speech works.
The second speech generates something that sounds like quite a low frequency mumble, and sounds nothing like 'lift'.

ntoll, 2017-03-08:
This has been fixed on master - pending a new release.

closed 2017-03-08 by ntoll

DavidWhaleMEF, 2017-03-08:
Thanks, although is it right to close the issue now, as it's not resolved for the user yet (me). Leaving it open will give visibility to others that it is a problem, and stop them logging the same issues.

DavidWhaleMEF, 2017-03-08:
@ntoll can you please provide a reference here to an issue number or a PR that fixes this issue?

ntoll, 2017-03-08:
Fixed via #345 :-)

ntoll, 2017-03-08:
You're welcome... thanks for the prod. :-)

ZanderBrown, 2017-03-08:
I'm sorry i don't want to add to people's inbox but it is quite amusing to see a comment timed '15 seconds from now', okay keeping on topic now i promise :-)