bbcmicrobit/micropython #421: micro:bit freezes when using radio AND pwm

kiesermeder, 2017-03-25:
I'm trying to realize a wireless pwm motor control and I'm stuck with a serious issue: The micro:bit freezes after an arbitrary time span (roughly one to twenty seconds) when using radio AND pwm.
It can then only be woken up again by reset button or power interruption. Along with the freeze it scrambles the content of the display.
When radio is on, one single execution of pinX.write_analog() is enough to activate the "time bomb", i.e. the micro:bit freezes shortly after.
I performed several tests to isolate the issue, all giving the result that it is related to pwm output:
- Tried pwm output at pin 0, 1, 2 and 8: same behaviour
- Tried different pwm cycles using pinX.set_analog_period(): same behaviour
- Tried coding with mu-0.9.13.linux.bin and online editor: same behaviour
- Tried different parameters of radio using radio.config(): same behaviour

At the moment I've no idea what can be done to avoid the freeze.

dpgeorge, 2017-03-25:
This is most likely a duplicate of #392.

@kiesermeder please read through that issue and see if you can fix your issue by using the latest firmware (details in that thread).

kiesermeder, 2017-03-26:
Thanks **dpgeorge**, after studying #392 I managed to use [http://micropython.org/resources/firmware/microbit-micropython.hex] provided by you. However, replacing firmware.hex in the PythonEditor didn't work for me for whatever reasons, so I had to download micropython, extract makecombinedhex.py and hexlifyscript.py from the tools/ folder and - as I've python3 installed on my system - invoke 'python3 makecombinedhex.py microbit-micropython.hex mycode.py > mycode.hex', and finally dragging the mycode.hex to the micro:bit.
Now radio and pwm are working without freeze.

DavidWhaleMEF, 2017-03-26:
So this proves that the fix on master is actually working? The final remaining step is to get master deployed into Mu on all platforms (including Raspberry Pi) and into the python.microbit.org hosted editor, which I know is already on @dpgeorge 's todo list. Thanks for confirming that the fix works @kiesermeder

DavidWhaleMEF, 2017-03-26:
@dpgeorge Do you know which github ticket we can refer to that captures the task of 'getting this deployed into all the editors' so we can link these tickets to it? Thanks!

DavidWhaleMEF, 2017-03-26:
We easily get 1-2 support tickets a week, about the PWM&Radio not working together. It would be good to have something that we can just point people at, so that they can monitor the status of this task themselves directly. Thanks

dpgeorge, 2017-03-28:
> 'getting this deployed into all the editors' so we can link these tickets to it

I don't think there is an issue for that, at least not in this repo.  And it's not really a development issue because as far as this repo is concerned the bug is fixed.  For each of the editors that host hex files, they would each need a separate "bug" report about updating to the latest hex file, but the problem with that approach is that it's an ongoing issue, that the hex will always need updating.

dpgeorge, 2017-03-28:
@DavidWhaleMEF sorry I can't be more helpful here.  I don't control the updating of hex files on any platform.

tim1mw, 2017-04-27:
I've had this problem as well and can confirm that with the official hex, the code crashes within a few seconds. I have compiled my own hex from a recent checkout of the master branch and incorporated it into my own fork of uflash (see https://github.com/tim1mw/uflash ). This is definitely better, but the PWM output usually goes off after about 5 to 10 minutes, although micro:bit doesn't actually crash and other functions continue to work, so that might be a separate problem.

ZanderBrown, 2017-04-27:
@tim1mw, to repeat comments by @ntoll, why fork when you can use `uflash -r firmware.hex` or `uflash --runtime=firmware.hex`

tim1mw, 2017-05-20:
ZanderBrown: I was trying to make life easy for anybody else who wanted a working uflash for PWM+Radio. Setting up the build environment to make the new firmware.hex file was a pretty painful experience, it seemed to me the best way to make this available to others was to make a new version of uflash that has it built in.

dpgeorge: The code is available here: https://github.com/tim1mw/Train-Bit/blob/master/src/recieve.py . You'll see that I've added some "ugly hacks" to try and counteract the PWM instability by rebooting on a regular basis, this helps a bit but I still have to use the manual reset button far more often than I would like. The companion transmitter code is stable.

tim1mw, 2018-01-30:
Any progress on this? I gained the impression from @DavidWhaleMEF when we spoke at PyCon UK 2017 that that it was being worked on.

DavidWhaleMEF, 2018-01-31:
@jaustin has been in contact with @dpgeorge and I believe that the testing is now in progress of the work that is about to be merged to master. Although getting that built into the various packaged forms (mu uflash and python.microbit.org) needs to happen before the general populus can access this work in beta form. It might already have made it into python.microbit.og/beta but @carlosperate can confirm that. If so, that might be a simple way for you to re-test your script?

dpgeorge, 2018-03-07:
Should be fixed in merge of version1 branch at e26d7c89d4a96de0fa0a1dd5aec024b31fc4816e

closed 2018-03-07 by dpgeorge