bbcmicrobit/micropython #423: ETX on UART causes KeyboardInterrupt

GreenSlime96, 2017-04-02:
Hi all,

I am developing a protocol on top of UART, and right now whenever the character ETX (ASCII 0x03) is sent, the REPL seems to activate after receiving a KeyboardInterrupt.

I am unable to catch the keyboard interrupt and it seems as if there's nothing I can do as the REPL is still running in the background.

Probably a bug as this issue doesn't happen on my PyBoards.

Many thanks,
Boon.

whaleygeek, 2017-04-02:
Check with @ntoll about using the raw REPL as he uses that for binary file
transfer in the Mu editor

dpgeorge, 2017-04-03:
There is currently no way around this: the underlying UART driver looks for a ctrl-C character in the UART interrupt handler (whenever a new character comes in) and converts it into a KeyboardInterrupt.

To make it possible to let these characters through would require a new method in the UART class to disable this behaviour, eg (as pyboard does it with USB VCP):
```python
uart.set_interrupt(-1) # don't capture ctrl-C characters
uart.read(10)
# etc
```
The pitfall with this is that you won't be able to interrupt a running script if you disable ctrl-C, but that's just something the users need to understand when using this advanced feature.

GreenSlime96, 2017-04-04:
Thank you for the clarification @dpgeorge, I looked through the documentation and source and wasn't able to find anything too helpful :)

dpgeorge, 2017-04-21:
This was fixed upstream: the micropython.kbd_intr() function was added.  But it will be some time before it filters down into this repo.

dpgeorge, 2018-03-07:
micropython.kbd_intr() was made available in 9af7add02c2769c2fc8c34caff782319e2c19661

closed 2018-03-07 by dpgeorge