bbcmicrobit/micropython #426: uart.write

Hartosan, 2017-04-04:
Hi,
When I execute this code, I get an error when executing uart.write.  The error is "TypeError: object with buffer protocol required"

any help would be appreciated.

from microbit import *
uart.init(baudrate=9600, bits=8, parity=None, stop=1,  tx=pin0, rx=pin1)
request = ""
while True:
    if button_a.is_pressed():
        uart.write("hello world")
        uart.write("\n")
    if button_b.is_pressed():
        request = uart.readinto(request)
        uart.write(request)

I

deshipu, 2017-04-04:
Have you read http://microbit-micropython.readthedocs.io/en/latest/uart.html#microbit.uart.uart.readinto ?
You can't write an integer with uart.write, you have convert it to a string.
But I think that your code should fail much earlier, as soon as it gets to uart.readinto -- you can't call that on a string, you need a mutable type.

Hartosan, 2017-04-13:
ah! thanks, newbie error.

closed 2017-04-21 by dpgeorge