bbcmicrobit/micropython #429: bluetooth for micropython??

KimHwaMok, 2017-04-05:
I know that micropython does not support Bluetooth due to memory issue.
But is there really no way about support Bluetooth?

If a program written in micropython has a structure linking all libraries, it seems to be able to save memory.
If it does, it seems to be able to support Bluetooth.
Am i wrong??

ukBaz, 2017-04-05:
I believe that @waynekeenan has done a proof of concept and has shared some details at:
https://thebubbleworks.com/bubblepy/

KimHwaMok, 2017-04-05:
@ukBaz Hi, thanks for reply.
So, you already use Bluetooth on micropython??

dpgeorge, 2017-04-05:
The Nordic soft device is 96k bytes code space (flash) for the basic S110 version and uses 8k of RAM (actually 6696 bytes for the absolute minimum case, don't know how useful that is though).  It also needs 1.5k of stack available to run, meaning you need that much extra space in the stack.  So in the best case you'd need to cut MicroPython down to 160k bytes of flash (that's easily doable but would mean you don't have speech or filesystem) and the heap would be about 6.5k (with the minimum BLE config and 3k stack).

KimHwaMok, 2017-04-06:
@dpgeorge Hi, again!
In my opinion, It seems to be more efficient if it supports Bluetooth even if it is excluding some functions. and It seems difficult to solve myself about support Bluetooth on micropython.

dpgeorge, 2017-04-07:
> In my opinion, It seems to be more efficient if it supports Bluetooth even if it is excluding some functions.

We cannot implement bluetooth in the standard build, and reduce the heap to 6.5k, because many users rely on having 9.5k of heap.  They also rely on all the other functions being there (eg music, filesystem).

Bluetooth may be possible as an alternative build of the code but that's a lot of work and I doubt anyone has time to do it at the moment.

closed 2017-04-11 by KimHwaMok

markshannon, 2017-04-17:
The bubblepy implementation is based on [nimBLE](https://mynewt.apache.org/v0_9_0/network/ble/ble_intro/) but like the softdevice and DAL, nimBLE is basically its own O/S which makes it difficult to integrate into micropython.
Why people have to write yet another framework instead of just writing a library, is a mystery to me :disappointed:

WayneKeenan, 2017-06-20:
The BubblePy 'static' flash & RAM usage stats are roughly:
190k Flash 	(70k for OS and BLE stack)
11.5K RAM   	(~1k BLE stack, ~2k OS, ~8.5k  MIcroPython,  3.5k )

Then ~3.5 k RAM is needed at runtime, so there's still 1k 'play'.