bbcmicrobit/micropython #430: translation(s) of documentation

JayBeRayBearGun, 2017-04-08:
Hello, 

I'd like to translate the microPython into Dutch. Naturall I would like to keep the translated version in sunch with the english language original source docs. 

What is the most efficient way to go about this?

regards,
JBR

deshipu, 2017-04-08:
You can look at the other started translations, and do the same:

https://github.com/techtabor/microbit-micropython-hu/
https://github.com/deshipu/microbit-micropython-pl/

Basically, you fork the github repository and do the translations, and then you periodically pull from the original English repository and rebase your translations on that of that. Merge conflicts will show you where you have new things to translate.

When you create your repository and build it on readthedocs, please let me know the name and I can then add it to the list of translations.

JayBeRayBearGun, 2017-04-14:
Ok then,.. not too familiar with the process yet but I understand it.

It wil be a while though before I upload something, so don't hold your breath.

closed 2017-04-14 by JayBeRayBearGun