bbcmicrobit/micropython #437: Image() buffer docs are confusing

DavidWhaleMEF, 2017-04-18:
microbit.org-ticket: 1447

The following docs about using the Image with a buffer are confusing, and actually, I can't get them to work at all. Am I missing something??!!

http://microbit-micropython.readthedocs.io/en/latest/image.html?highlight=Image

First of all, it is not clear that kwargs are not supported in the docs, users are trying this form and getting an error:

[image]

Secondly, the docs talk about passing in a buffer as an array of integers

[image]

Python does not have an array type, it uses lists? Do you mean a list?

I tried with two forms of lists, both give a TypeError: object with buffer protocol required

[image]

[image]

Can someone send me a working example of how this can be made to work? Thanks!

deshipu, 2017-04-18:
Python does have an array type: https://docs.python.org/3.6/library/array.html

dpgeorge, 2017-04-18:
> First of all, it is not clear that kwargs are not supported in the docs, users are trying this form and getting an error:

A lot of the built-in methods/functions in CPython don't allow keyword args, so we are not doing anything different here.  I guess it's just a case of make the docs explicit in this regard.  Maybe we should list the 4 differnt constructors for image, eg:
```python
Image() # create a blank 5x5 image
Image(s) # create an image by parsing the string (single char returns that glyph)
Image(w, h) # create a blank image of given size
Image(w, h, data) # create an image from the data
```

> Secondly, the docs talk about passing in a buffer as an array of integers
>  ...
> Can someone send me a working example of how this can be made to work?

You need to pass an object with the buffer protocol, something like a bytes object or bytearray will do it:
```python
Image(2, 2, b'\x00\x01\0x1\x00')
```
Note that the values you pass are raw bytes, so the above is like [[0,1],[1,0]].

DavidWhaleMEF, 2017-04-18:
Thanks. Can you add this example to the docs, and also explain the lack of kwargs support, as both are actually not that clear even to me until you gave me a worked example!

DavidWhaleMEF, 2017-04-18:
The docs say pass an array of integers - what you have above is an array of bytes?

DavidWhaleMEF, 2017-04-18:
I see Python (and MicroPython) do have an array, thanks @deshipu - nothing comes up in the docs when I search for array though??

[image]

DavidWhaleMEF, 2017-04-18:
I can confirm that this code works:

from microbit import *

i = Image(5,5,b'\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08\x08')

display.show(i)

If you could update the docs with an example, and also explain the fact that you can't use kwargs, that will be great, thanks for your help.

microbit-carlos, 2018-10-30:
Implemented in #582.

closed 2018-10-30 by microbit-carlos