bbcmicrobit/micropython #438: Bytes prepended to radio receive payload

DavidWhaleMEF, 2017-04-21:
I'm looking into a discrepancy between how payloads transmitted by PXT are received by MicroPython radio.receive_bytes() - see this ticket in PXT here for a repro case:

https://github.com/Microsoft/pxt/issues/1935

I see there is a 'spurious' 01 00 01 at the start of the payload received, which I cannot find documented anywhere in the official MicroPython docs here: https://microbit-micropython.readthedocs.io/en/latest/radio.html

However, I might have dreamed it, but I am sure I saw something in either a github ticket or on the MicroPython mailing list saying something about extra bytes prepended at the start of a received payload, but I can't find it.

(a) could you explain what the purpose of the 01 00 01 at the start of the received payload is

(b) If you can remember where this surfaced (github ticket or mailing list) can you point me to it, so I can confirm that I'm not going mad(!)

(c) Can we update the documentation to explain what these bytes are for and what they mean?

Thanks

DavidWhaleMEF, 2017-04-21:
For completeness, here is a packet trace...

So really, this payload as received by MicroPython should be interpreted thus:

```
01 00 01|02|80 AF 03 00|00 00 00 00|(05) 44 41 4C 45 4B
01 00 01           prefix added by MicroPython
02                 string
80 AF 03 00        time
00 00 00 00        serial number disabled
05                 length
44 41 4C 45 4B     DALEK
```

dpgeorge, 2017-04-21:
> I am sure I saw something in either a github ticket or on the MicroPython mailing list saying something about extra bytes prepended at the start of a received payload, but I can't find it.

Yes those bytes are added so to be compatible with the DAL.  Well, we were compatible with a version of the DAL when the code was written but maybe things have changed since then (in which case the DAL made a breaking change to itself and code from one version of the DAL may not be able to communicate with code from another version).

ntoll, 2017-04-21:
The docs you're looking for are here: https://microbit-micropython.readthedocs.io/en/latest/radio.html#radio.receive

DavidWhaleMEF, 2017-04-21:
Thanks @ntoll you confirmed that I wasn't dreaming this up and I am not going mad :)

I think this might be incorrect now actually, but best to leave everything as it is, otherwise we'll create too many variants of 'different' and never get any sense of interoperability between the various languages.

closed 2017-04-21 by DavidWhaleMEF

ntoll, 2017-04-21:
If this has changed, then, as @dpgeorge states, the DAL has introduced a breaking change. :-(

finneyj, 2017-04-21:
@DavidWhaleMEF, @dpgeorge, @ntoll 

No breaking changes to this on microbit-dal side. I think PXT folks may have added an additional optional wrapper however that includes some metadata, which could be the source of the issue.

@DavidWhaleMEF I knew that once a raw radio interface like this started getting used then folks would want to experiment with their own protocols, which is a good thing IMHO, so added three byte to (ironically) help manage interoperability. The three bytes are a protocol id (1 for basic raw datagram), version number (as defined by the protocol itself) and a group number. This is the 01 00 01 you're seeing there - it's a raw datagram version 1 frame in group 0.

ntoll, 2017-04-21:
@finneyj thanks for the very speedy update! :-)

If PXT change things, how do they expect others in the ecosystem to work with their platform? Surely this is something the MB Foundation can coordinate so we're all pulling together in this regard?

finneyj, 2017-04-21:
hey @ntoll - No worries. Nice to speak to you again. 

Having interfaces that don't interoperate between languages is is no-one's interest. I agree that some co-ordination is needed here, along with a nice healthy dollop of tolerance on the side. :-)

The way I see it, people are bound to experiment and develop their own things (in fact, if they didn't we would have failed spectacularly), so we need to be tolerant to people doing that - things evolve, grow and converge with time... They're rarely nailed the first time around.

Encouraging evolution to happen in such a way that it doesn't negatively affect others is the trick. This was the original idea behind having protocol numbers, so my starter for ten would be to maintain (and encourage deeper adoption) of that as a simple and established underlying mechanism and then have an open discussion about what protocols we define and support as standard for the microbit ecosystem. 

I'd be happy to start a discussion with my foundation hat over on microbit-dal about that if the relevant folks would be happy to engage.

dpgeorge, 2017-04-21:
Note that MicroPython supports sending (and receiving) arbitrary frames without any headers.  Just use the send_bytes and receive_bytes functions, then you should be able to interoperate with PXT.

finneyj, 2017-04-21:
@dpgeorge I think @DavidWhaleMEF is using that interface isn't he? The difference is one of transparency. If the job of filtering the header is left to the programmer then it's always going to look spurious unless they understand what is is...

ntoll, 2017-04-21:
@finneyj couldn't agree more WRT encouraging evolution and experimentation. The trick to making this a success is to communicate (hence my suggestion for MBF coordination) just so *everyone* is kept in the picture. :-)

DavidWhaleMEF, 2017-04-21:
Here is my contribution...

A simple message logger, this shows the raw bytes as received from PXT and was what I used to create the traces in this ticket above.

from microbit import *
import radio

radio.config(channel=7, address=0x75626974, group=132, data_rate=radio.RATE_1MBIT)
radio.on()

while True:
    sleep(500)
    try:
        msg = radio.receive_bytes()
        if msg is not None:
            line = ""
            for b in msg:
                line += str.format("%02X " % b)
            print(line)
    except Exception as e:
        # reset the radio if it crashes
        print("  RESET:%s" % str(e))
        radio.off()
        radio.on()

DavidWhaleMEF, 2017-04-21:
This was the code I wrote to strip strings out. This is clearly a bit wrong now, as the '1' at the start is part of @finneyj protocol header, really it should be looking at the '2' further down to identify it as a string. But it's a good starting point.

```
from microbit import *
import radio

radio.config(group=132)
radio.on()

def get_message():
    while True:
        try:
            msg = radio.receive_bytes()
            if msg is not None:
                if msg[0] == 1 and len(msg) > 13:
                    lstr = msg[12] # length byte
                    text = str(msg[13:13+lstr], 'utf-8')
                    return text

        except Exception as e: # reset radio on error
            print("reset %s" % str(e))
            radio.off()
            radio.on()

while True:
    print("waiting...")
    display.show('?')

    m = get_message()
    print('got message:', m)
    display.scroll(m)
```

DavidWhaleMEF, 2017-04-21:
There is currently nothing in our official documentation about this protocol header from what I can see? @finneyj let's work up a k-base article together on this as a starting point, explaining both the design and current implementation, and also creating in that a registry of protocol indexes and formats that we can update as people register new protocols with us. I'll create a placeholder and we can work it up together.

dpgeorge, 2017-04-21:
Just to be clear and answer the title of this issue: MicroPython does not add anything to a received packet.  radio.receive_bytes() returns the exact bytes found on-air in the payload.  radio.receive() will check that the first 3 bytes are 01 00 01 and then strip them, returning the rest as a str.

DavidWhaleMEF, 2017-04-21:
Thanks @dpgeorge makes sense. 

If we use radio_receive() with binary data, will the conversion to str() ever fall over, and is it possible to transparently index into characters and interpret their ordinal value? In which case, that might be a neater way to receive and process these payloads.

Thanks

DavidWhaleMEF, 2017-04-21:
It's rough and ready, but here is the beginnings of a document store where things like this can be published. It might not be it's permanent home, but it's a start. I'll think through the collaboration aspects of this in a week or two (probably just a github repo hosted at github.com/microbit-foundation I suspect).

https://support.microbit.org/solution/articles/19000053363-mbtn2-protocol-headers

finneyj, 2017-04-21:
understood - thanks for clarification @dpgeorge.

dpgeorge, 2017-04-22:
> If we use radio_receive() with binary data, will the conversion to str() ever fall over

Yes, the conversion will (or at least should in principle) fail if the bytes are not a properly encoded utf8 stream.  That's standard Python.

> is it possible to transparently index into characters and interpret their ordinal value?

Again, this is just standard Python: indexing into bytes object will return the ordinal (0-255), indexing into a str object returns a str of length 1 (which could be implemented behind the scenes as multiple *bytes* due to utf8 encoding).