bbcmicrobit/micropython #439: PIN mode changes are overly restrictive?

plummet555, 2017-04-23:
Hi

My issue results from an attempt to play music on a PIN other than PIN 0 (preferably on PIN13). Using the tip runtime code cloned yesterday it fails like this:

>>> music.play(music.FUNK,pin13)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin 13 in spi mode

Trying to write analog fails similarly:

>>> pin13.set_analog_period(500)
>>> pin13.write_analog(500)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin 13 in spi mode

Looking through the code I can see that is because pinmode_error is mapped as the 'release' function for SPI pins (as well as others) so attempts to acquire the pin for other purposes always fail;

const microbit_pinmode_t microbit_pinmodes[] = {
    [MODE_UNUSED]        = { MP_QSTR_unused, noop },
    [MODE_WRITE_ANALOG]  = { MP_QSTR_write_analog, analog_release },
    [MODE_READ_DIGITAL]  = { MP_QSTR_read_digital, noop },
    [MODE_WRITE_DIGITAL] = { MP_QSTR_write_digital, noop },
    [MODE_DISPLAY]       = { MP_QSTR_display, pinmode_error },
    [MODE_BUTTON]        = { MP_QSTR_button, pinmode_error },
    [MODE_MUSIC]         = { MP_QSTR_music, pinmode_error },
    [MODE_AUDIO_PLAY]    = { MP_QSTR_audio, noop },
    [MODE_TOUCH]         = { MP_QSTR_touch, pinmode_error },
    [MODE_I2C]           = { MP_QSTR_i2c, pinmode_error },
    [MODE_SPI]           = { MP_QSTR_spi, pinmode_error }
};

But - with the block editor I can successfully write analog to PIN13. So, this does not appear to be a hardware limitation.

I think that if SPI is not being used then it should be possible to release PIN13 and to use it for other purposes?

Related to this, does anyone have a more accurate description or schematic of the pins? The diagrams show that (e.g.) PIN13 is SPI1/SCK and PIN11 is button B, but that is just their primary functions. I think that any pin can be used for digital I/O and most (all?) can be used for PWM?

Thanks

DavidWhaleMEF, 2017-04-23:
There is a detailed data sheet of the pins on our tech site here, with all alternative functions documented: http://tech.microbit.org/hardware/edgeconnector_ds/

plummet555, 2017-04-23:
Thanks - that's exactly what I was looking for.

I think that confirms that the limitation is a bug as PIN13 is shown as supporting PWM.

To prove the point, I changed those pinmode_error entries to noop and I'm able to play music and/or output PWM on PIN13

ntoll, 2017-04-23:
I believe this was fixed in current master... @dpgeorge @markshannon ..?

plummet555, 2017-04-24:
No, I tested on the latest master s/w and you can still see the pinmode error mappings for (e.g.) the SPI PINs here:

https://github.com/bbcmicrobit/micropython/blob/master/source/microbit/microbitpinmode.c

ntoll, 2017-04-24:
Hmmm, my bad. I mis-remembered.

I seem to remember I changed the lines in microbitpin.cpp (https://github.com/bbcmicrobit/micropython/blob/master/source/microbit/microbitpin.cpp#L53) so the SPI and I2C pins started in `MODE_UNUSED`.

I believe subsequent conversations with @dpgeorge and @markshannon led me to the conclusion that something more permanent than my hacky fix had resolved the problem.

If the SPI and I2C pins cannot be used for something other than SPI or I2C then we've made a breaking change to the pin behaviours. :-(

My question is why can't pins start in unused mode and then, if SPI or I2C is used, they acquire the correct mode..?

markshannon, 2017-04-24:
Merging https://github.com/bbcmicrobit/micropython/pull/407 should fix the SPI case.
The I2C pins are always used for the accelerometer and compass.

ntoll, 2017-04-24:
Thank you @markshannon! I knew I hadn't been dreaming.

So @plummet555, the answer is that there's a fix in the works. It needs reviewing / testing before it'll land in master. Care to take a look and give it a drive..?

plummet555, 2017-04-24:
Thanks @ntoll and @markshannon 

I tried with the  [#407](https://github.com/bbcmicrobit/micropython/pull/407)  change (cherry-picked on top of latest master) and it does indeed fix the issue for the SPI pins:

>>> pin13.get_mode()
'unused'
>>> music.play(music.FUNK,pin13)

Let's review the other pins though - specifically those where the mode is automatically defined on reset and where the release function is mapped to pinmode_error (meaning they cannot be assigned to other uses such as digital I/O):

MODE_DISPLAY: Pins 4,6,7,9,10
>> Per 'Uncoupling Default Functionality' in the edge connector [document] (http://tech.microbit.org/hardware/edgeconnector_ds/), the screen can be disabled allowing the pins to be reused (subject to some limitations) so I think ideally the code would allow these pins to be released if the screen has been disabled?

MODE_BUTTON: Pins 5,11
>> Similarly, although there are pull-up resistors it should be possible to re-use these pins

MODE_I2C: Pins 19,20
>> As @markshannon pointed out there are components physically tied to these lines so I'm inclined to agree that it should not be possible to release these pins. Interestingly though the 'Uncoupling Default Functionality' section stops short of saying 'you can't do it' so maybe this shouldn't be prevented by the MicroPython code either?

plummet555, 2017-04-24:
..d'oh, I just thought to try display.off() and it already releases the pins so ignore that part of the comment

dpgeorge, 2018-03-07:
Fixed by e04006d67764da0e12251f9621d3254b7fe9b23a

closed 2018-03-07 by dpgeorge