bbcmicrobit/micropython #440: Problems appending script to recent runtime

bmustill-rose, 2017-04-24:
I am experiencing issues when flashing a device with a runtime based on the latest commit ( a11ce23). I'm using uFlash to append my scripts & flash devices but I don't believe uFlash to be the issue.

* I have compiled an unmodified runtime of MicroPython using the above referenced commit.
* Flashing my runtime to the device without a script and carrying out some brief tests suggests that it's working properly - E.G. display, buttons, speech and radio all work.
* Using uFlash to flash any script and supplying my runtime results in the device seemingly doing nothing. The script is not executed, nothing is printed to serial & I am unable to access the REPL.
* If I flash my runtime, rename my script to main.py and "put" it on the device using microfs it works - E.G. the script executes on boot, behaves as I would expect it to & I'm able to access the REPL.

Please see the attached for a selection of files that demonstrate the problem. For what it's worth my runtime was compiled under Windows 10 X64 which I realize might not be a typical build environment.

Thanks.
[files.zip](https://github.com/bbcmicrobit/micropython/files/951562/files.zip)

bmustill-rose, 2017-04-24:
See https://github.com/ntoll/uflash/issues/38 for some previous discussion.

ZanderBrown, 2017-04-24:
Im assuming `hello.py` is the code your trying to flash, `runtime.hex` is what you're adding to and `combined.hex` is the result.

First observation:

`runtime.hex` is bigger than `combined.hex`.
Either i've misseinterpreted the names or somehow adding to the file has reduced it's size

carlosperate, 2017-04-24:
Difference in size is due to the line endings, runtime.hex has windows CRLF, while combined contains the LF-only endings.

markshannon, 2017-04-30:
The problem is simply that the runtime is now too big. It exceeds 248k and the script gets stored at 248k.

https://github.com/bbcmicrobit/micropython/pull/434 reduces the size of the runtime to less that 215k.

See https://github.com/ntoll/uflash/issues/38 for a possible temporary fix on the µflash side.

microbit-carlos, 2018-03-16:
The new `version1` branch merged into master should have reduced the runtime to expected limits, so this shouldn't be an issue anymore.
Apart from that the latest uFlash already contains a version of the v1.0b and seems to be working correctly.
I'm closing this one, but feel free to reopen if the problem persists or reappears. Thanks for the report!

closed 2018-03-16 by microbit-carlos