bbcmicrobit/micropython #441: micropython/docs/display.rst has /*in params for scrol

tabroughton, 2017-04-25:
As a basic user I am copying and pasting the example in the docs and I get an error because of this asterisk, is the asterisk useful and is there another approach I should be taking?

whaleygeek, 2017-04-25:
Unfortunately, what you have pasted there is not an example call to the scroll function, it is a formal specification of the format of the parameters.

[image]

The docs on this site: http://microbit-micropython.readthedocs.io/en/latest/display.html?highlight=display are generated using an automated documentation tool, and it generates this formal parameter specification automatically.

I'll leave it up to someone like @ntoll or @dpgeorge to explain precisely what that '*' actually means. I've tried to find a definitive guide to parameter passing in Python on the web for you, but failed to find a good description, Nicholas or Damien are most likely to know where to find the official docs that explain how the python formal parameter passing syntax works.

However, some typical calls of this scroll function would be something like:

```
from microbit import *

display.scroll("hello")
display.scroll("hello", delay=100)
display.scroll("hello", wait=True, loop=True)
```

ntoll, 2017-04-25:
If this is causing confusion we should perhaps make the documentation clearer.

However, @whaleygeek is correct about *why* you see this in documentation. The document that specifies this is here: https://www.python.org/dev/peps/pep-3102/

What does it mean? Simply that all the arguments that follow the `*` must be named if you pass them, rather than using the position of the argument to specify the value to pass into the function. I.e. you MUST say `wait=True` rather than just pass in `True`.

Hope this makes sense.

tabroughton, 2017-04-27:
Thanks for the explanation.  I do think it's unclear for folks who might want to find examples of how to use the api and aren't aware of this.  Maybe, given the level of developer knowledge of users/students that the micro:bit is targeted at and that this is the main source of how to use the API, providing an alternative view that can be copied and pasted to try things out would be useful.

deshipu, 2017-04-27:
The official documentation is targeted at experienced people, who can then take this information and present it in form of prepared lessons, tutorials, experiments, etc. more suitable for teaching. The reason for that is that we have very limited resources, and documenting things in *any* way is the top priority. Then, if there is need and resources, that base documentation can be expanded upon and made more friendly.

tabroughton, 2017-04-27:
yes, makes sense when you put it like that.

closed 2017-04-27 by tabroughton

whaleygeek, 2017-04-27:
@ntoll @jaustin Just a note, I know we spoke a while ago about the possibility of some more target-audience focused resources for MicroPython on the micro:bit, as these existing (excellent) resources are clearly developer focused. 

Are there any other members of the huge worldwide Python community that any of us know, that we could reach out to, to get some help writing some target-audience (kids and teachers) focused reference documentation? The groundwork is done already on the developer docs, so it is as you say, a development from these, rather than a clean sheet. But feedback we are getting from target users is that the docs are not very accessible given their specific levels of expertise with Python.

ntoll, 2017-04-27:
I would want to know:

* What's wrong with the tutorials? (Why don't they appeal to beginner developers or teachers?)
* What's wrong with the API docs?

Please give specific examples. For example, "Mr.Jones told me he didn't understand the singing example in the tutorial because of X, Y and Z reasons." It's the X, Y and Z that we need to find, written down, by the users who encountered the problems.

DavidWhaleMEF, 2017-04-27:
It would be a little time consuming, but when we get some more resource at this end, I could commission a trawl through our 1600 support tickets to find people that have specific python problems, and try and make you a bit of a list. This could be valuable user research for us all. However, for data protection reasons, we can't share specific tickets with you.

From the top of my head, and having done a bit of a scan through tickets, the key issues are:

1. **Parameter specs are too formal**, what people need is real examples of code they can cut and paste that works. 

I suspect the issue here to be a combination of: not all methods in the API have usages in the examples; the API reference does not link through each API method to a tutorial that shows it in use; there is not a one or two liner example against all of the API methods in the reference (those that do have a short usage example are really good)

I think the fix is to provide a link from every API method in the reference, to a tutorial that uses that API method, and provide a short 1 or 2 line example in the reference docs of calling the function with a sensible example set of parameters that they can cut and paste to start from. My gold standard here is the Waite's Group Turbo C Bible from the 1990's that I learnt to code in C from when I was a student.

It also seems to me that the differentiation between tutorials and reference is not immediately obvious (except for a bit of nav on the left and the technical tone & presentation shifting up a gear or two).  I wonder if we could brand a tutorial page slightly different from a reference page, so that users know when they are in the 'playful area' and when they are in the 'serious stuff' areas?

We see a number of scripts where people have cut and paste the formal parameter spec into their code, then say they can't get 'the examples in the documentation' to actually work. This happens when there is a '*' in the parameter list, of course.

2. Users stray into **aspects about Python that are not documented in these docs**, and their only next step is to go to the big wide internet and look for general Python docs. Those are very formal and often show examples using Python features that is not supported inside the micro:bit+MicroPython combination, so they get side tracked into something that then won't work on the micro:bit. 

Yes, its a full implementation of the Python 3 language, but a huge amount of the standard library for good reasons is not included. Users don't really know the difference between what is the language and what is the standard library, so they can't make that judgement themselves. Perhaps we can find a way to help them differentiate between the two so they can self help a bit more?

It's not completely clear to me what the solution to this is. However, my 1980's BBC micro manual listed all the features of the language in one place, and that was key to me being able to find and learn about all the features quickly when I was a kid. 

We get a number of tickets from users requesting a 'list of all commands supported by the micro:bit when in Python mode' - we send them to these docs and they say 'that isn't quite what I meant, there are lots of words in the language I see in example programs that are not in those docs, please send us a list of all the commands we can use on the micro:bit'. If we send them to official Python docs, they come back and say that 'I was reading about this feature xyz but when I tried it on the micro:bit it didn't work and gave an error, it turns out it is not supported, please give us a list of all *supported* commands on the micro:bit when in python mode'.

We also sometimes see users stray into MicroPython (generic) documentation, and then try something out on the micro:bit and it doesn't work that way 'oh, I thought micro:bit ran MicroPython, is that not the case?' for example.

3. Some users tell us that when they see **'warning this is alpha code'** they tend to shy away from it (e.g. the speech module) 

Is this still alpha code or is it now tested and working? Could we just remove the alpha messaging here? Is it going to be changing to Beta soon, Release Candidate, or is it actually released and all working now?

4. **What does this error code mean** - users get a specific exception message and don't know how to interpret the meaning of that. We get lots of requests from users asking for a list of all the error codes (in this case a mix of panic codes and Python exceptions - we have panic codes on our kbase). 

The best that came back from the community here was to 'look at the standard Python docs', but see point 2, what users want to see is the list of all error codes (exceptions) that the micro:bit version can generate, and a description (presumably with an example) of why that exception might be generated, and things they can do to try to fix it.

5. **Concurrency** - we get a number of tickets where users don't understand, or want help with, making more than one thing happen at the same time. They send us scripts where they have got into a real pickle and need to start again basically.

The issue here is we have users migrating from their visual experience in PXT which supports event handling, to a textual experience in Python that does not, and they seem to really struggle with applying their learning from PXT into Python, in the same way that users moving from Scratch to Python struggle a bit here.

[... cut ...]
I think the fix here is to make sure that each page in the system that describes a top level concept (e.g. audio) does actually have navigation in the nav menu on the left, so that users can find it from browsing.

7. **Ranges on parameters** some are not documented

This one catches me out too. what is the valid range of frequencies you can pass to the music.pitch? What is the valid range of BPM you can send to the music module - i.e. what are the extents of the designed in parameters of the system?

I'm sure with proper user research effort we could surface better information, but this is the best I can produce at the moment, I hope it helps a bit?

ntoll, 2017-04-27:
@whaleygeek an epic and amazingly useful list. This is *exactly* the sort of thing I was hoping for.

However, while we should support users I'd caution against giving them all the answers. Rather, I'd prefer documents that encouraged playful exploration and confident questioning. If they read the tutorials they'll gain enough knowledge and explanation about much of the stuff you mention to be competent MicroPython programmers who know what sort of things to ask.

DavidWhaleMEF, 2017-04-27:
Thanks :)

Regarding the caution, I hear what you say and support it. But when you are learning something new for the first time, and trying to teach kids some stuff in a 1 hour lesson, there just isn't the time to read everything. The curriculum is jam packed as you know. I'm personally very much against 'copy me' resources as there is very little learning.

So I think the best compromise is to remove any big holes where we can, signpost to relevant nuggetised reading where we can, and make sure that any accuracy problems we find along the way are fixed. That will get us close, I think.

ntoll, 2017-04-27:
I believe we're in the same page (if not paragraph or even sentence) in this respect. :-)

DavidWhaleMEF, 2017-04-27:
Yes, the cursor is lined up to the pixel, Nicholas :)