bbcmicrobit/micropython #444: strange behaviour show vs scroll

barrachri, 2017-05-07:
I am trying 2 programs having the microbit with the face up:

```python
from microbit import *

while True:
    gesture = accelerometer.current_gesture()
    if gesture == "face up":
        print('face up')
        display.show("S")
    else:
        print('face down')
        display.show("N")
```
It works correctly, showing `S`.

But with this code I just get the `N` scrolling (leaving the microbit in the same position)
```python
from microbit import *

while True:
    gesture = accelerometer.current_gesture()
    if gesture == "face up":
        print('face up')
        display.scroll("S")
    else:
        print('face down')
        display.scroll("N")
```

what's wrong with me or with the `scroll`?

Version of MicroPython: `MicroPython v1.7-9-gbe020eb on 2016-04-18; micro:bit with nRF51822`

dpgeorge, 2017-05-10:
The gesture detection requires a few (about 10) readings of the accelerometer before it can "decide" what the new gesture is.  So it will stay stuck in the existing gesture until you make about 10 calls to any of the accelerometer methods.

Your first program runs very fast so you don't notice this effect.  Your second program takes about 2 seconds to scroll the message on the display and so you notice the effect.

You can try something like this:
```
last_gesture = ''
while True:
    gesture = accelerometer.current_gesture()
    if gesture != last_gesture:
        last_gesture = gesture
        if gesture == "face up":
            print('face up')
            display.scroll("S", wait=False, loop=True)
        else:
            print('face down')
            display.scroll("N", wait=False, loop=True)
```

barrachri, 2017-05-10:
Thanks @dpgeorge 👍

closed 2017-05-10 by barrachri