bbcmicrobit/micropython #447: round() does not always work

nielsswinkels, 2017-05-26:
In my code where I print out the calculated speed of a vehicle passing 2 points (triggered by touch on pin 0 and 1), the round function does not always work. Sometimes it shows correctly 2 decimals, sometimes it shows 1 decimal, and sometimes it shows a lot of decimals.

```
from microbit import *
distance = 25

while True:
    display.show(Image("00000:"
              "00000:"
              "00000:"
              "00000:"
              "90009"))
    while pin0.is_touched() == False:
        pass
    firstGate = running_time()
    display.show(Image("90000:"
              "90000:"
              "90000:"
              "90000:"
              "90009"))
    
    while pin1.is_touched() == False:
        pass
    secondGate = running_time()
    display.show(Image("90009:"
              "90009:"
              "90009:"
              "90009:"
              "90009"))
    sleep(500)
    speed_ms = (distance/100.0)/((secondGate-firstGate)/1000.0)
    speed_kmh = speed_ms * 3.6
    display.scroll(str(round(speed_kmh, 2)))
```

But if I try the minimal code it does seem to always work:
```
from microbit import *

while True:
    display.scroll(str(round(2.123456, 2)))
```

dpgeorge, 2017-05-26:
This looks like a duplicate of #367.  To work around the issue please use `"%.2f" % speed_kmh`, instead of `str(round(...))`

closed 2017-05-26 by dpgeorge

DavidWhaleMEF, 2017-05-26:
If this is a common question people are asking, would it make sense to update the docs so they could discover this themselves?

I've done this in our support knowledge-base, but it would be much better to surface a warning about the limited resolution of the micro:bit MicroPython close to where people will be using it, which is in the docs.

Here is an article I wrote the last time this was logged as an issue:

https://support.microbit.org/solution/articles/19000026098-strange-rounding-error

dpgeorge, 2017-05-27:
> If this is a common question people are asking, would it make sense to update the docs so they could discover this themselves?

That would be nice, but in fact the bug is fixed upstream and it would be a better solution to just get that bug fix in this repo and in a microbit release.