bbcmicrobit/micropython #448: Gestures do not work

DavidWhaleMEF, 2017-06-05:
The following code:

```
from microbit import *

while True:
    print(accelerometer.get_gestures())
    sleep(1000)
```

Always outputs
()
()
()

Regardless of whether I shake the microb:bit, make it face up, down, etc etc.

Traceability: Mu version: https://codewith.mu/help/0.9.13/

DavidWhaleMEF, 2017-06-05:
Also for completeness, I tried this with two different micro:bits, and most of the time (shaking the thing madly) I get False:

```
from microbit import *

while True:
    print(accelerometer.was_gesture("shake"))
    sleep(250)
```

DavidWhaleMEF, 2017-06-05:
I tried this on many micro:bits now, all with the same effects.

Also, I discovered that if you change the delay in the loop, the effectiveness of the 'was' sensing is slightly increased.

from microbit import *

```
while True:
    if accelerometer.was_gesture("shake"):
        print("shake")
    sleep(100)
```

The most effective response was when the sleep was removed alltogether. However, this doesn't tally with the description in the docs, which states:

microbit.accelerometer.was_gesture(name)
Return True or False to indicate if the named gesture was active since the last call.

dpgeorge, 2017-06-06:
Gestures are not updated in the background but rather there needs to be constant calls to some accelerometer method to do the gesture detection.  Please see #214 for the same/similar question asked before.

If you make the `sleep(1000)` in your first example into a `sleep(50)` does it work?

DavidWhaleMEF, 2017-06-06:
Oh, that makes sense. So you're co-operatively scheduling, and leaving the scheduling up to the user.

Thanks. I'll try later.

Would it be possible to update the documentation to state something to that effect, as it wasn't at all clear from the docs - the docs just say 'update since the last call' - which is strictly true, but unless the reader understands that you have to call this very often in order for it to work, it looks like it doesn't work. Also could you advise in the documentation how often the method should be called (on what time period) for the algorithm to perform comparably to the other languages where this self-scheduling is not required?

Thanks

hackscribble, 2017-06-06:
Hi @DavidWhaleMEF @dpgeorge hope you don't mind me jumping in.

I've made some hand-timed measurements using the following code:

```
from microbit import *

last_gesture = ""

while True:
    gesture = accelerometer.current_gesture()
    # sleep(50)
    sleep(100)
    # sleep(200)
    if gesture is not last_gesture:
        last_gesture = gesture
        print('>{g:s}<'.format(g=gesture))
```
With `sleep(200)`, the time between changing the orientation of the micro:bit and seeing the new gesture printed was about 2.4 seconds.  With `sleep(100)` it was about 1.2 seconds and pro rata for `sleep(50)`.

So it looks like it takes 12 readings of the accelerometer for a change of gesture to be recognised.

smartwombat, 2018-09-18:
I have the same issue using the example morse.py from make.techwillsaveus.com/microbit/activities/micro-morse ...
MU 1.0.0 Windows 10, although unless it's a library bug the OS shouldn't matter ?

DavidWhaleMEF, 2018-09-18:
@smartwombat The problem you see in MU 1.0.0 was introduced accidentally as part of a change made in the lower level DAL software here: https://github.com/lancaster-university/microbit-dal/issues/372

A new MicroPython (and Mu) release is being baked at present I think to mop up issues like this - @carlosperate has been looking into this.

jaustin, 2018-10-03:
What do we want to put in the docs here? Just a note that says 'If the gestures aren't firing, make sure you have the latest version of MicroPython'? or similar? @carlosperate

microbit-rosslowe, 2018-10-04:
I think a note on https://github.com/bbcmicrobit/micropython/issues/448#issuecomment-306434228 possibly in https://microbit-micropython.readthedocs.io/en/latest/tutorials/gestures.html?highlight=gesture or https://microbit-micropython.readthedocs.io/en/latest/accelerometer.html?highlight=gesture :

> Gestures are not updated in the background but rather there needs to be constant calls to some accelerometer method to do the gesture detection.

And maybe a word on using gestures in a loop with a smaller `sleep()` delay. (See PR / commit above)

microbit-carlos, 2018-10-17:
Note has been added to PR #569 and it has been merged to master.
Thank you all for your input! If you think it still needs to be further clarified, please feel free to reopen or create a new issue.

closed 2018-10-17 by microbit-carlos