bbcmicrobit/micropython #454: using the radio to talk to raspberry pi and more

swfiua, 2017-06-19:
it would be good to use bluetooth radio on other devices to talk to the radio.

For instance if there was a raspberry pi module that was as easy to use as microbit.radio and the two could have conversations it would enable some really interesting projects.

is anyone working on something like this already?

deshipu, 2017-06-19:
I spent some time trying to get `microbit.radio` to talk with the popular and cheap `nrf24L01` module -- this should be in theory possible, as they both use the same packet protocol. However, I've given up on that after a few days -- for some reason I was only able to receive 1 message in 10000. I would need better tools to be able to figure out what is happening there -- I even arranged to borrow an SDR radio capable of listening on that frequency, but in the end I didn't, and I shelved the project.

Note that there are relatively cheap USB dongles with NRF51 chip on them -- the same chip as on the microbit -- that could be programmed to be a bridge between the computer and the microbit -- but for now the easiest way to do that is to simply use a second microbit connected via usb.

DavidWhaleMEF, 2017-06-19:
Bluetooth and radio are very different things.

By far the cheapest solution is just use a second micro:bit plugged into the Raspberry Pi. And as @deshipu comments, it's probably much much less hassle as well. And the micro:bits are really easy to get hold of, unlike some of the other platforms.

I have various projects that solve part of your problem, and am currently merging them into one project (bitio) which will soon allow you to 'import radio' and then just chat away with the micro:bit radio network. But in the interim, the microbit-gateway is what I use on a lot of my projects

https://github.com/whaleygeek/microbit-gateway

https://github.com/whaleygeek/bitio

swfiua, 2017-06-19:
Thanks.  I took old father timr for a walk to the lighthouse...

And tgen the usb penny dropped.

Thanks for the awesome answers.

https://github.com/whaleygeek/microbit-gateway

https://github.com/whaleygeek/bitio

rhubarbdog, 2018-08-19:
I wanted to control a micro:bit robot with a joystick, so I linked my raspberry pi to a microbit using uart. I then wrote a uart radio relay to send a simple message to the robot code to link the microbit to a raspberry pi is covered <a href="https://github.com/rhubarbdog/uart-producer-consumer">here</a>.
The radio relay is just a modified version of `microbit.consumer.py` that does `radio.send(message)` rather than `display.scroll(message)`

DavidWhaleMEF, 2018-08-20:
Nice!

I often use this as a basis for multi-micro:bit projects with radio...

https://github.com/whaleygeek/microbit-gateway