bbcmicrobit/micropython #458: No basic.forever style of loop in Micropython

OrangutanGaming, 2017-07-18:
In the Javascript version, there is a loop that can make it so you can run other things at the same time, where as Micropython doesn't have a feature like this which can cause some features to not respond properly due to other tasks being run at the same time.

dpgeorge, 2017-07-19:
> In the Javascript version, there is a loop that can make it so you can run other things at the same time, where as Micropython doesn't have a feature like this

This was a deliberate design decision (to not have this feature), to keep the execution model simple enough to understand.  But you can achieve a similar thing using the following simple pattern:
```python
while True:
    do_single_iteration_of_forever_code()
    do_other_processing()
```
True multithreading capabilites are beyound the scope/capabilities of the microbit.

> which can cause some features to not respond properly due to other tasks being run at the same time

Do you have an example of something that doesn't "respond properly"?  Things like display.scroll and music.play can run in the background.

OrangutanGaming, 2017-07-19:
Well I actually just started and converted the pong game that's taught in the JavaScript editor lessons into Python, and when I try to move the paddle, sometimes the input isn't always used, I believe due to it doing other tasks. Here is the entire code: 
```python
import microbit

ballY = 2
ballX = 1
paddleY = 2
ballXVelocity = 1
ballYVelocity = 1
gameRunning = False

microbit.display.set_pixel(ballX, ballY, 9)
microbit.display.set_pixel(0, paddleY, 9)
microbit.display.set_pixel(0, paddleY + 1, 9)

def paddleNotUp():
    return paddleY > 0

def paddleNotDown():
    return paddleY < 4

def ballOutOfRange():
    return ballX <= 0

def gameOver():
    global gameRunning
    gameRunning = False
    microbit.display.clear()
    microbit.display.scroll("GAME OVER", 150)

def updateXVelocity():
    if ballX == 4:
        global ballXVelocity
        ballXVelocity = (-1) * ballXVelocity
    elif ballX == 1 and ballY == paddleY:
        global ballXVelocity, score
        ballXVelocity = (-1) * ballXVelocity

def updateYVelocity():
    if ballY == 4 or ballY == 0:
        global ballYVelocity
        ballYVelocity = (-1) * ballYVelocity

def moveBall():
    global ballX, ballY
    microbit.display.set_pixel(ballX, ballY, 0)
    ballX = ballX + ballXVelocity
    ballY = ballY + ballYVelocity
    microbit.display.set_pixel(ballX, ballY, 9)

while True:
    if microbit.button_a.is_pressed():
        gameRunning = True
        if paddleNotUp():
            microbit.display.set_pixel(0, paddleY, 0)
            paddleY = paddleY - 1
            microbit.display.set_pixel(0, paddleY, 9)

    elif microbit.button_b.is_pressed():
        gameRunning = True
        if paddleNotDown():
            microbit.display.set_pixel(0, paddleY, 0)
            paddleY = paddleY + 1
            microbit.display.set_pixel(0, paddleY, 9)

    if gameRunning:
        if ballOutOfRange():
            gameOver()
        else:
            updateXVelocity()
            updateYVelocity()
            moveBall()
            microbit.sleep(400)
```

ntoll, 2017-07-20:
Use `button_a.was_pressed()` instead of `is_pressed`. See https://microbit-micropython.readthedocs.io/en/latest/button.html#Button.was_pressed

I suspect that might be the sort of behaviour you're looking for. Right?

OrangutanGaming, 2017-07-22:
That has helped, however it still isn't as responsive as the `basic.forever` method

hackscribble, 2017-07-22:
Hi @OrangutanGaming 

You could try replacing the blocking `sleep(400)` call with a non-blocking timer based on `running_time()`. This should speed up responsiveness to button presses while still only updating velocities and moving the ball every 400ms (which is what sets the speed of the game, I guess).

Try something like this (not tested).  

```
<snip>

while True:
    if microbit.button_a.was_pressed():
        gameRunning = True
        last_update_time = 0  # timer will expire immediately the first time it is tested below
        if paddleNotUp():
            microbit.display.set_pixel(0, paddleY, 0)
            paddleY = paddleY - 1
            microbit.display.set_pixel(0, paddleY, 9)

    elif microbit.button_b.was_pressed():
        gameRunning = True
        last_update_time = 0  # timer will expire immediately the first time it is tested below
        if paddleNotDown():
            microbit.display.set_pixel(0, paddleY, 0)
            paddleY = paddleY + 1
            microbit.display.set_pixel(0, paddleY, 9)

    if gameRunning:
        if ballOutOfRange():
            gameOver()
        elif running_time() - last_update_time > 400:
            last_update_time = running_time()  # "reset" the timer
            updateXVelocity()
            updateYVelocity()
            moveBall()
```

OrangutanGaming, 2017-07-23:
This works well, but the problem is, when you press a button, it speeds up the ball making it skip pixels.

hackscribble, 2017-07-23:
Fixed that problem.  Also included two alternative statements (one commented out) in each of the blocks of "game start" code to control how quickly you want the ball to start moving after the button is pressed to begin game.
 
```
import microbit

UPDATE_PERIOD = 400

ballY = 2
ballX = 1
paddleY = 2
ballXVelocity = 1
ballYVelocity = 1
gameRunning = False

microbit.display.set_pixel(ballX, ballY, 9)
microbit.display.set_pixel(0, paddleY, 9)
# microbit.display.set_pixel(0, paddleY + 1, 9)

def paddleNotUp():
    return paddleY > 0

def paddleNotDown():
    return paddleY < 4

def ballOutOfRange():
    return ballX <= 0

def gameOver():
    global gameRunning
    gameRunning = False
    microbit.display.clear()
    microbit.display.scroll("GAME OVER", 150)

def updateXVelocity():
    if ballX == 4:
        global ballXVelocity
        ballXVelocity = (-1) * ballXVelocity
    elif ballX == 1 and ballY == paddleY:
        global ballXVelocity, score
        ballXVelocity = (-1) * ballXVelocity

def updateYVelocity():
    if ballY == 4 or ballY == 0:
        global ballYVelocity
        ballYVelocity = (-1) * ballYVelocity

def moveBall():
    global ballX, ballY
    microbit.display.set_pixel(ballX, ballY, 0)
    ballX = ballX + ballXVelocity
    ballY = ballY + ballYVelocity
    microbit.display.set_pixel(ballX, ballY, 9)

while True:
    if microbit.button_a.was_pressed():
        if not gameRunning:
            gameRunning = True
            # last_update_time = 0  # timer will expire immediately the first time it is tested below
            last_update_time = microbit.running_time()  #  time will not expire until UPDATE_PERIOD after game start
        if paddleNotUp():
            microbit.display.set_pixel(0, paddleY, 0)
            paddleY = paddleY - 1
            microbit.display.set_pixel(0, paddleY, 9)

    elif microbit.button_b.was_pressed():
        if not gameRunning:
            gameRunning = True
            # last_update_time = 0  # timer will expire immediately the first time it is tested below
            last_update_time = microbit.running_time()  #  time will not expire until UPDATE_PERIOD after game start
        if paddleNotDown():
            microbit.display.set_pixel(0, paddleY, 0)
            paddleY = paddleY + 1
            microbit.display.set_pixel(0, paddleY, 9)

    if gameRunning:
        if ballOutOfRange():
            gameOver()
        elif microbit.running_time() - last_update_time > UPDATE_PERIOD:
            last_update_time = microbit.running_time()  # "reset" the timer
            updateXVelocity()
            updateYVelocity()
            moveBall()
```

OrangutanGaming, 2017-07-24:
Works perfectly!

closed 2017-07-24 by OrangutanGaming

ntoll, 2017-07-24:
Hurrah! :+1: :tada: