bbcmicrobit/micropython #459: Misleading wiring diagram for NeoPixels

DavidWhaleMEF, 2017-07-25:
This image is slightly misleading...

https://github.com/bbcmicrobit/micropython/blob/master/docs/neopixel-croc.png

It shows a banana plug labelled 5V connected to the 3V pad. 

This may give the impression to users that they can back-power the micro:bit from 5V using this connector (which is definitely not the case and damage will occur if they do).

The data sheet labels it as VDD

https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf

[image]

Also note, depending on which specific variant of the ws2812 you get, the supply voltage is out of spec if powered directly from a micro:bit - although in practice the lower voltage version does actually work regardless.

But I think that labelling the wire 5V is misleading and could lead to some damage if mis-interpreted by users.

carlosperate, 2017-07-25:
I agree that VDD is probably a better name. As you note here, it would also be helpful to add that the voltage provided by the micro:bit is slightly lower than that required by the nexopixels (same goes with servos, most 5v ones should work, but not all do) and the fact that a battery powered micro:bit will drop the provided voltage as the battery drains.

The diagram was made with https://pycomic.github.io/microbit.html , but this web app does not offer labels for the pin, so we might have to edit the image anyway.

ntoll, 2017-07-25:
@carlosperate that's precisely what I did (used the GIMP to add labels). If it would be handy, it's relatively easy to add text to an SVG such as the one used at the website.

Neon22, 2017-07-25:
if it is svg then inkscape will do the job more effectively IMHO.
- https://inkscape.org

jaustin, 2018-10-17:
How about this:
[image]
I'll also upload the SVG to the docs with the commit so that people can generate these labelled images based on the source too:
[neopixel-croc.zip](https://github.com/bbcmicrobit/micropython/files/2486683/neopixel-croc.zip

DavidWhaleMEF, 2018-10-17:
LGTM

Do we need to signpost the potential voltage mismatch to users in some way (either here or on tech site, or as a kbase article?) We had a lot of chatter and mis-information in the various communities about this, and the problem is worse if you run the micro:bit off of batteries due to the depressed cell voltage, and differently worse if running off USB due to the lower advertised available current. Perhaps a kbase would be better for that issue?

jaustin, 2018-10-17:
Yea, I'll factor that into the docs when I change the image over and we can discuss in review in a PR. Gives us a nice chance to talk about blue LED 'on' voltages and batteries discharging which I think fits nicely with the style of the MicroPython docs where they tell you about adjacent/useful things as well as just the API.

microbit-carlos, 2018-10-31:
PR #596 has been merged, thank you everyone!

closed 2018-10-31 by microbit-carlos