bbcmicrobit/micropython #460: Missing license text in header files

glennrub, 2017-08-17:
Would it be possible to get an update on the headers in micropython/inc/microbit/ by the authors? 

Right now there is no permission to use, modify or redistribute these headers as far as i understand it. An update would be much appreciated.

DavidWhaleMEF, 2017-08-17:
There is a blanket LICENCE which presumably covers the whole repo, here: https://github.com/bbcmicrobit/micropython/blob/master/LICENSE

@dpgeorge or @ntoll can you comment on why the headers in all the other folders have a full MIT licence at the top, and these one's don't? Was it just an oversight?

DavidWhaleMEF, 2017-08-17:
Or @carlosperate might have a view on this.

glennrub, 2017-08-17:
A bit of background; I would like to cherry-pick microbitdisplay.h and microbitimage.h moving them out of repo context, modify and redistribute. The corresponding source files have a license that allows me to do this.

dpgeorge, 2017-09-14:
Sorry for the slow response... indeed the LICENSE file in the root of the repo covers those files without an explicit header; see #193 and #229 for background discussion on this point.

That said, it would be good to update all files to have a license header.

glennrub, 2017-10-04:
Got it. For now I'll copy the LICENSE and AUTHORS files and put these files along with the files missing explicit license text.