bbcmicrobit/micropython #461: bug in read_analog

lgnativs, 2017-08-23:
the code in fucntion microbit_pin_read_analog is:
`mp_obj_t microbit_pin_read_analog(mp_obj_t self_in) {
    microbit_pin_obj_t *self = (microbit_pin_obj_t*)self_in;
    microbit_obj_pin_acquire(self, microbit_pin_mode_unused);
    nrf_gpio_cfg_input(self->name,  NRF_GPIO_PIN_NOPULL);   // if pin is output mode,this will change to correct mode
    analogin_t obj;
    analogin_init(&obj, (PinName)self->name);
    int val = analogin_read_u16(&obj);
    NRF_ADC->ENABLE = ADC_ENABLE_ENABLE_Disabled;
    return mp_obj_new_int(val);
}`

deshipu, 2017-08-23:
Yes?

lgnativs, 2017-08-23:
after set set pin on write_digital/write_analog mode,GPIO of nRF51822 is config as output state. the function of microbit_pin_read_analog didn't change the pin of GPIO to input mode, because adc channel must to be floating input.

dpgeorge, 2017-08-29:
On the current version (50c41dee2791d1521e793132232951da37fdee65) read_analog() works ok for me on pin0.  It works fine from a clean boot, after doing pin0.write_digital() (writing either 0 or 1) and after doing pin0.write_analog().

@lgnativs can you please provide a (minimal) script/code which doesn't work as expected, as well as the version you use, found by running `import os` followed by `os.uname()`.

dpgeorge, 2018-03-07:
I assume this was resolved.

closed 2018-03-07 by dpgeorge