bbcmicrobit/micropython #463: set_pull AttributeError

DavidWhaleMEF, 2017-08-24:
set_pull as documented in the published docs, does not work...

http://microbit-micropython.readthedocs.io/en/latest/pin.html?highlight=set_pull

[image]

[image]

Mu version 0.9.13

DavidWhaleMEF, 2017-08-24:
FYI @jaustin @carlosperate @ntoll any comments? What is the URL to the correct version of the MicroPython docs for the current latest Mu 0.9.13 and for the version deployed at python.microbit.org (i.e. where the docs match the deployed codebase)? Thanks!

carlosperate, 2017-08-24:
There is no URL as Read The Docs has only got the "latest" version building from master.

Tracking back a little we have Mu last updated uFlash in https://github.com/mu-editor/mu/commit/39876bfad8775d685b95632583997adde3b40b16 and uFlash updated the hex file in https://github.com/ntoll/uflash/commit/fe26fe8ea0b7943674cb8165e6939cba3ef0dc88 on the 23rd of October 2016. Assuming it was a build from master, then we are looking at https://github.com/bbcmicrobit/micropython/commit/525d88c73f2b81d319efa3889cf5424ebfdc1770 .

If we create a tag to that commit (0.9 is the original micro:bit march release, so 0.10? 0.9.13 to correlate with Mu?), it looks like Read The Docs could pick it up? I'm not entirely sure: http://docs.readthedocs.io/en/latest/versions.html#how-we-envision-versions-working
Otherwise we have a branch already for 0.9, we could make one for this version as well.

I think @deshipu has some experience configuring these types of docs as well, so it'd be great to get your input.

For reference, we have these issues discussing versioning: https://github.com/bbcmicrobit/micropython/issues/370 and https://github.com/bbcmicrobit/micropython/issues/199

deshipu, 2017-08-24:
We can enable any of the tags or branches to be build by readthedocs,
and we can also make any of them the default. They will appear in the
menu in the bottom left corner, where the green "v: latest" is, and you
can switch between them (or link to them).

carlosperate, 2017-08-24:
Thanks @deshipu, that's great. Can all that be done from within the Read the docs panel? Ideally that'd be the best way as to avoid having to change any of the files in the repository.

deshipu, 2017-08-24:
Yes, see https://readthedocs.org/dashboard/microbit-micropython/versions/

carlosperate, 2017-08-24:
I think only owners have access to that page (I get a 404), but it sounds like the easiest way forward is to create a branch with the "new" version name for https://github.com/bbcmicrobit/micropython/commit/525d88c73f2b81d319efa3889cf5424ebfdc1770 and then add it as a version in Read The Docs.

carlosperate, 2017-08-24:
Btw, what branch/commit is the "stable" version of the docs based on?

deshipu, 2017-08-24:
Hmm, there is no such branch in the repository now, but maybe there used to be?

carlosperate, 2017-08-24:
Based on http://docs.readthedocs.io/en/latest/versions.html it's probably meant to point to the last tag, probably the same as 0.9 now?

DavidWhaleMEF, 2017-12-13:
@ntoll FYI re: https://twitter.com/Vlastimil_Hovan/status/940884073446367233

DavidWhaleMEF, 2017-12-13:
@carlosperate is it possible to fix this do you know (get the docs in line with the currently deployed version of MicroPython in mu 0.9.13 and python.microbit.org) - I often see people bumping into this particular issue with set_pull, and I don't know what other aspects of the docs are out of step with reality to know if people are struggling with other things.

carlosperate, 2017-12-13:
It looks like we already have `v0.9` of the docs: http://microbit-micropython.readthedocs.io/en/0.9/

We should probably set it as the default for http://microbit-micropython.readthedocs.io instead of `latest`, until v1.0 is released. I don't have access to this project, so cannot see what commit does `0.9` points to, or change any of the settings. @ntoll or @deshipu would be able to do that.

deshipu, 2017-12-13:
I just changed the default to be 0.9.

carlosperate, 2017-12-13:
Thanks @deshipu!

DavidWhaleMEF, 2017-12-13:
Thanks @deshipu and @carlosperate !!

DavidWhaleMEF, 2017-12-13:
The workaround here (Reported by Vlastimil Hovan on twitter here: https://twitter.com/Vlastimil_Hovan/status/940902464575950848 is...

use analog_read()
and make sure you are aware that all 3 round pins have a weak pull-up fitted on the board.

The technical data sheet is here: http://tech.microbit.org/hardware/edgeconnector_ds/

The weak pullups (used for touch sensing) are 10M pullup (and internally the pins by default will most likely have a 12k on-chip pull-down enabled)

microbit-carlos, 2018-03-08:
https://github.com/bbcmicrobit/micropython/issues/498 has been created to make sure we have the correct default docs until v1.0 is released.

closed 2018-03-08 by microbit-carlos