bbcmicrobit/micropython #464: Published docs are out of date with the deployed code version

DavidWhaleMEF, 2017-08-24:
The version of the docs published are not in step with the version of the codebase currently deployed.

https://codewith.mu/ gives me 0.9.13 of Mu, it's not clear to me which version of MicroPython-microbit is built into that package as I can find no traceability.

The docs here http://microbit-micropython.readthedocs.io/en/latest/ describe features that are not available in the currently deployed codebase within codewith.mu or python.microbit.org

See: https://github.com/bbcmicrobit/micropython/issues/463

The micro:python API is moving to version 1.0, but this message seems to have been there for months?

[image]

Speech is still ALPHA code?

[image]

DavidWhaleMEF, 2017-08-24:
@jaustin is this in hand, do you know? Thanks.

lgnativs, 2017-08-24:
I have the same problems.

microbit-carlos, 2018-03-08:
The master version of the docs (for v1.0) should then remove these warnings.

microbit-carlos, 2018-10-16:
Both of these are covered in PR #567.

nealmcb, 2019-10-10:
Are there any docs still out-of-date with MicroPython 1.0.0 release? Or should this be closed so the 1.0.o milestone can be cleaned up....

microbit-carlos, 2019-10-10:
The docs are now in sync, thanks for bringing this up @nealmcb! I'll close this issue :)

closed 2019-10-10 by microbit-carlos