bbcmicrobit/micropython #465: Bug in write_analog

SimonShapiro, 2017-08-27:
I think there is problem with write_analog.

I have a potentiometer wired to the micro:bit and the readings from the potentiometer are sent to a pin attached to an LED in order to control the brightness of the LED. 

The following one line of micro:bit javascript works, but the same line in microPython does not.

**The javascript code**

        pins.analogWritePin(AnalogPin.P1, pins.analogReadPin(AnalogPin.P0))

which is the result of the block code works fine and the brightness of the LED changes as the potentiometer is changed.

**The microPython code does not change the intensity of the LED**

    pin1.write_analog(pin0.read_analog())

does not change the intensity of the LED.

Any ideas of what is going on here?

DavidWhaleMEF, 2017-08-27:
It would be worth checking what the default pulse width period is for write_analog in MicroPython, as it may be different to that in the Javascript Blocks editor. I can't see this documented anywhere in the official docs, so you might want to set it to a known value so you know what it is.

If the analog period is too high or too low, you risk not getting enough energy into the LED and it won't be visible.

You could try adding this at the top of your program, before your loop:

http://microbit-micropython.readthedocs.io/en/latest/pin.html?highlight=write_analog#microbit.MicroBitAnalogDigitalPin.set_analog_period

Also, do you have any form of loop delay in your loop (please provide your full program listing), as repeatedly calling write_analog in a loop without delays may be repeatedly loading the PWM peripheral such that it never actually gets to output a value as it is always being reconfigured.

microbit-carlos, 2018-03-13:
I've tested this with the following code, changing the pin0 read values simply by touching the pin, and using a voltmeter to read the average voltage from pin1:

```python
from microbit import *

while True:
    read_value = pin0.read_analog()
    pin1.write_analog(read_value)
    print(read_value)
```

I can confirm this was an issue with v0.9 of MicroPython, where the voltage read was close to 0 even when pin0 was reading values at around 80-90% ADC range. 
Tested on v1.0 (version1 branch merged into master at e26d7c89d4a96de0fa0a1dd5aec024b31fc4816e) and this version has fixed this issue.

So I'll close this one, but feel free to reopen if you still have any questions.

closed 2018-03-13 by microbit-carlos