bbcmicrobit/micropython #466: lightLevel()

mobluse, 2017-09-05:
https://makecode.microbit.org/ has lightLevel() function that does return light level. It would be great i MicroPython for Microbit also had this. The theory behind lightLevel is here: https://support.microbit.org/support/solutions/articles/19000024023-how-does-the-light-sensing-feature-on-the-micro-bit-work-

DavidWhaleMEF, 2017-09-05:
The light level is coded into the device abstraction layer (DAL), but MicroPython does not use much of the DAL in it's codebase for various reasons, so it doesn't get access to all of the same features as all of the other languages. 

@markshannon did a bit of work on a branch somewhere to look into adding this, but I think he has been quite busy since.

I did look into adding this myself a while ago, but it was quite a distraction from the actual project I was doing that needed it, and I focused my time on the application (which I wrote in PXT) instead of spending a lot of time adding one feature.

Mark might be able to share the branch he was working on here perhaps, if others in the community wanted to pick it up and finish the work off.

mobluse, 2017-09-05:
Thanks for your information about DAL:
https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitLightSensor.cpp
Maybe one could do the same in MicroPython without using the C++ library.

ntoll, 2017-09-06:
Work to make this happen was done at PyCon UK 2016 by Zeth and I seem to remember seeing it working. If you're at this year's PyCon UK you should find Zeth and see if he can push the work he did.

DavidWhaleMEF, 2018-02-16:
@dpgeorge  I heard a mention that micro:bit/MicroPython V1 is now built on top of the DAL V2. If this is the case, can we enable the light sensing feature please?

This is in order to make the facilities available in the various coding languages more consistent. It's easier for everyone to be able to write resources that work on both makecode and micro:bit/MicroPython if the feature sets of the languages are compatible, and the net effect of this is that there will be more MicroPython resources available too, which is a good thing! (I personally have a number of great light sensing resources that would become available in Python too if this feature was enabled)

dpgeorge, 2018-02-18:
> I heard a mention that micro:bit/MicroPython V1 is now built on top of the DAL V2.

Yes that's true.

> If this is the case, can we enable the light sensing feature please?

Unfortunately it's not as easy as just enabling it.  MicroPython still uses its own display driver so would need to do special things to cooperate with the DAL's light sensor.  I'll try to find a way to do it.

microbit-carlos, 2018-07-16:
`display.read_light_level()` method added in https://github.com/bbcmicrobit/micropython/commit/02ef4d03bef9a72f60bb7f0f2ae1b90c75cd9432, thanks Damien! 🎉

closed 2018-07-16 by microbit-carlos

DavidWhaleMEF, 2018-10-27:
FYI @carlosperate a user asking if this has been added, as it's not in docs? https://twitter.com/MrDBurrin/status/1056160737725501440

microbit-mark, 2018-10-29:
@DavidWhaleMEF  https://microbit-micropython.readthedocs.io/en/latest/display.html#microbit.display.read_light_level