bbcmicrobit/micropython #467: Reading out any Python errors for blind kids instead of (or as well as) scrolling them on the LEDs

dumbledad, 2017-09-08:
I’ve been working recently on a tangible system for teaching visually impaired kids to code called [Torino](https://blogs.microsoft.com/ai/2017/03/15/project-torino-microsoft-creates-physical-programming-language-inclusive-visually-impaired-children/).  One of the blind kids is ready to progress to a text based language. 

His school mates are using Python on the micro:bit and so one of our collaborators has come up with a series of micro:bit Python lessons for him. 

But he was stuck on the very first lesson:

from microbit import *
import speech
speech.say("hello")

He could write the program in Notepad++ using his JAWS screen reader. He could flash it to his micro:bit using the keyboard shortcut to uflash that we set up. He has the amp:bit to drive headphones or speakers. But no sound came out. Eventually his mum and I deciphered the scrolling text on the micro:bit, it said “error: unexpected indentation”. Without any of us noticing, his entire program was indented one tab stop. We removed the tabs and the program started working.

He is blind and so there is no way he could access that useful error message (in fact it was a struggle for his mum and I).

That got me thinking. It would be really useful for blind users if the micro:bit had the facility to read out its error messages instead of (or as well as) scrolling them. Is that a plausible patch to write? Any pointers on where I might start?

ntoll, 2017-09-08:
@markshannon thoughts?

deshipu, 2017-09-08:
I think the easiest and most flexible solution would be to use the screen reader on the REPL.

ntoll, 2017-09-08:
@deshipu aha! That's a *really* great idea - I wish I'd thought of that. ;-)

@dumbledad connecting to the REPL is *very easy* and means your student can use native tools and a speech synthesiser that doesn't sound like a drunk dalek talking through a pillow case. :-) I'd be more than happy to help with making this happen (I might be in Cambridge next week visiting the Raspberry Pi Foundation if you're up for a coffee and a chat sometime, once I know the exact day?).

dumbledad, 2017-09-08:
@ntoll Definitely - let's do that. Email me on timregan at microsoft dot com and we'll schedule

Does the screen reader on the REPL mean that we are moving him away from the micro:bit? If so I am a bit nervous about that as my goal is not just to help him learn how to code, but also the help him learn how to code with his school friends. They are using Python on the micro:bit.

dumbledad, 2017-09-08:
@SaqibS may have thoughts too

armbennett, 2017-09-08:
The REPL is great for live coding and experimentation, however when writing longer and more complex programs it can become limiting. It is very powerful for students to be able to flash a program to the micro:bit, unplug it and experience it running totally independently from the computer, especially when working with the accelerometer.

deshipu, 2017-09-08:
REPL is not only useful for experimentation, it also displays the output of your programs and detailed information about the errors -- much better than what scrolls on the display. Of course you can unplug your microbit at any time once you finished debugging.

dumbledad, 2017-09-08:
@deshipu Does that mean we can write a Python micro:bit program and run it first in the REPL to monitor any error conditions and then, when it runs, flash it to the micro:bit?

ntoll, 2017-09-08:
@armbennett completely agree with you about the REPL and working unplugged.

However, I also agree with @deshipu - during the development of a script you can `print()` things so they appear in the REPL. This is *great* for debugging. Also, all exceptions and errors are also reported on the REPL. Once your script is finished *remove* the `print` statements and work unplugged.

Does this make sense?

ntoll, 2017-09-08:
@dumbledad pretty much yes... please see my immediately previous reply.

armbennett, 2017-09-08:
@ntoll that makes sense. Can it be used without access to the command line as the laptop he is using is provided by the school he isn't allowed access to the command line which has caused problems in the past. Many other visually impaired children will be in a similar position.

ntoll, 2017-09-08:
On Windows, assuming you have set up the Windows driver for USB serial, you should use putty.exe (http://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) to connect to the board. Use Window’s Device Manager to discover the COM port to which the PyBoard is connected. Within the putty application, click Session in the left hand panel, click the Serial radio button, enter the COM port discovered via the Device Manager in the Serial Line box, and then click the “open” button.

Is this helpful?

armbennett, 2017-09-08:
Yes that’s a good idea, thanks!

dumbledad, 2017-09-08:
Interesting, and then we'd be see the REPL on the device over serial via PuTTY?

ntoll, 2017-09-08:
Got it in one. :-)