bbcmicrobit/micropython #469: error building version 1

carlosperate, 2017-09-15:
Ubuntu version:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.1 LTS
Release:        16.04
Codename:       xenial
```

GCC ARM version:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Useful bits from build output:
```
$ yt clean
$ yt up
info: get versions for bbc-microbit-classic-gcc-nosd
info: get versions for mbed-gcc
info: get versions for microbit-dal
info: update outdated: microbit-dal@2.0.0-rc9 -> microbit-dal@v2.0.0-rc10 from GitHub lancaster-university/microbit-dal
info: download microbit-dal@v2.0.0-rc10 from GitHub lancaster-university/microbit-dal
info: get versions for mbed-classic
info: update outdated: mbed-classic@0.0.4 -> mbed-classic@microbit_hfclk+mb6 from GitHub lancaster-university/mbed-classic
info: download mbed-classic@microbit_hfclk+mb6 from GitHub lancaster-university/mbed-classic
info: get versions for ble
info: update outdated: ble@2.5.0 -> ble@v2.5.0+mb3 from GitHub lancaster-university/BLE_API
info: download ble@v2.5.0+mb3 from GitHub lancaster-university/BLE_API
info: get versions for ble-nrf51822
info: update outdated: ble-nrf51822@2.5.0 -> ble-nrf51822@v2.5.0+mb7 from GitHub lancaster-university/nRF51822
info: download ble-nrf51822@v2.5.0+mb7 from GitHub lancaster-university/nRF51822
info: get versions for nrf51-sdk
info: update outdated: nrf51-sdk@2.2.0 -> nrf51-sdk@v2.2.0+mb4 from GitHub lancaster-university/nrf51-sdk
info: download nrf51-sdk@v2.2.0+mb4 from GitHub lancaster-university/nrf51-sdk
```
```
$ yt build
info: generate for target: bbc-microbit-classic-gcc-nosd 0.2.3 at /home/vagrant/vagrant_shared/micropython/yotta_targets/bbc-microbit-classic-gcc-nosd
GCC version is: 6.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
suppressing warnings from ble-nrf51822
suppressing warnings from nrf51-sdk
suppressing ALL warnings from mbed-classic, ble, ble-nrf51822 & nrf51-sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /home/vagrant/vagrant_shared/micropython/build/bbc-microbit-classic-gcc-nosd

...

[191/317] Building C object source/CMakeFiles/microbit-micropython.dir/home/vagrant/vagrant_shared/micropython/source/py/objset.c.o
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```
Same problem using the makefile.
Master branch builds correctly in this environment.

[full_build_op.txt](https://github.com/bbcmicrobit/micropython/files/1306691/build_op.txt)

dpgeorge, 2017-09-15:
Is your filesystem case insensitive?  The problem is that there are some include files that only differ by case, eg inc/microbit/microbitdisplay.h vs yotta_modules/microbit-dal/inc/drivers/MicroBitDisplay.h.

This can be fixed by being more explicit in the include and using a path, eg: `#include "microbit/microbitdisplay.h".

carlosperate, 2017-09-15:
Yes, I am building in a shared folder with Windows and it looks like this makes the directory case insensitive.
```
$ touch caseSensitive
$ ls -l caseSensitive
-rwxrwxrwx 1 vagrant vagrant 0 Sep 15 15:16 caseSensitive
$ ls -l casesensitive
-rwxrwxrwx 1 vagrant vagrant 0 Sep 15 15:16 casesensitive
```
I assume this also means it's currently not possible to build on Windows ?
@finneyj you might be interested to read this as well.

I'll try again in a case sensitive environment.

dpgeorge, 2017-09-15:
Thanks for confirming.  Should be fixed by 8c3c9afe5c7b680abc7a76a501ec973c5ca0477e

carlosperate, 2017-09-15:
I was able to build correctly in a case sensitive filesystem, thanks Damien!

On the case senstive Windows shared folder I got a bit further, but still encountered errors: 
[build_op_2.txt](https://github.com/bbcmicrobit/micropython/files/1306959/build_op_2.txt)

dpgeorge, 2017-09-17:
Ok, should be fully fixed by e20e41797cb56e94907e0122981d352ac28e973c

carlosperate, 2017-09-19:
Thank you @dpgeorge! Can confirm it compiles now in my case insensitive environment.

closed 2017-09-19 by carlosperate