bbcmicrobit/micropython #471: Micropython can not work when powered by the battery.

ouki-wang, 2017-09-26:
I have compiled  Micropython firmware but found nrf51822 cannot work when powered by the battery;
however, it won't happen when used firmware wrote by others. 

What caused the problem?

ouki-wang, 2017-09-26:
Why connect TGT_NRESET to p0.19 of nrf51822?
I could not find corresponding codes.

dpgeorge, 2017-09-26:
> I have compiled Micropython firmware but found nrf51822 cannot work when powered by the battery;
> however, it won't happen when used firmware wrote by others.

You'll need to give more detail about your problem: what board you use, which firmware version of MicroPython, what script you are using, which "other" firmware is working, and what exactly are the symptoms when it doesn't work.

> Why connect TGT_NRESET to p0.19 of nrf51822?
> I could not find corresponding codes.

Please explain in more detail your issue/question.

ouki-wang, 2017-09-28:
My source code commit id is 50c41dee2791d1521e793132232951da37fdee65 which has been compiled to firmware A.

The other firmware: the built-in mu-1.9.13 version as firmware B.

Compared with firmware B, only the firmware A could not work well when powered by the battery. 

Below is the firmware A phenomenon:

1. When power-supplied by USB, TGT_NRESET voltage is 3.1V which nrf51822 recognized as a high level. 

2. When  power-supplied by 3.7V lithium battery, TGT_NRESET voltage is 2.0V while nrf51822 would not recognize it as a high level.

3. You can use Micropython when comment source code related to resetButton.

After preliminarily analyze, I guess 2.0V cannot been recognized as a high level voltage. Moreover, the soft reset bottom should not been comment. How could I solve this problem?

dpgeorge, 2017-09-28:
@ouki-wang thanks for the details about the problem.

> You can use Micropython when comment source code related to resetButton.

Where is the code related to the reset button that you refer to here?

Can you please try using "version1" branch of this repository, the commit c8c4db8444fa35cda2bbca94b703be7469b666fe, and see if that works when running from a battery?

ouki-wang, 2017-09-28:
@dpgeorge thanks for your reply
MicrobitDisplay.cpp
                //burn cycles
                while(i>0)
                {
                    // Check if the reset button has been pressed. Interrupts are disabled, so the normal method can't be relied upon...
                    if (resetButton == 0)
                        //microbit_reset();//--------------here----------

                    i--;
                }

MicrobitSupperMain.cpp
int main()
{    
    // Bring up soft reset button.
    //resetButton.mode(PullUp); // -----------here--------------
    //resetButton.fall(microbit_reset);// -----------here--------------

dpgeorge, 2017-09-28:
What hardware are you using, is it an official microbit?  Did you check that your battery is not flat?  Did you plug it into the battery connector and what is the voltage at the VDD pin of the nRF51822?

ouki-wang, 2017-09-28:
hardware: official microbit V1.3B
battery is good,battery voltage is 3.9V
When power-supplied by lithium battery only,  nrf51822 VDD=3.8V, TGT_NRESET=2.2V

dpgeorge, 2017-09-28:
Are you able to try the version1 branch, commit c8c4db8444fa35cda2bbca94b703be7469b666fe?

ouki-wang, 2017-09-28:
@dpgeorge  ok，I will try this branch。but I connect overseas sites very slow when a lot of people online。
so I need a long time to give you a reply

ouki-wang, 2017-09-28:
@dpgeorge version1 branch-commit id c8c4db8 work well

dpgeorge, 2017-09-28:
> version1 branch-commit id c8c4db8 work well

Great, thanks for confirming that.  That branch will become the master at some point.  It's recommended to use it now if you want to build your own firmware.

microbit-carlos, 2017-11-28:
>hardware: official microbit V1.3B
battery is good,battery voltage is 3.9V
When power-supplied by lithium battery only, nrf51822 VDD=3.8V, TGT_NRESET=2.2V

@ouki-wang just as a general note, the maximum operating voltage of the nrf51 is 3.6V, anything above that can cause damage and the microcontroller might not work to specifications, and the recommended voltage for battery connector or edge connector is 3.0V.

That being said, glad this does work on master. Sounds like this issue can be closed.

microbit-carlos, 2018-03-15:
`version1` branch has been merged to master in e26d7c89d4a96de0fa0a1dd5aec024b31fc4816e, so we can close this one. Thanks for the report and confirming it works with the latest versions.

closed 2018-03-15 by microbit-carlos