bbcmicrobit/micropython #474: First two transactions after i2c.init() raise OSError -1010

deshipu, 2017-10-15:
I'm writing a driver for controlling the SH1106 OLED display over I²C from micro:bit, and I'm using pins 1 and 2 for this, because of their easy connectivity. I noticed that after I call `microbit.i2c.init(freq=400000, scl=microbit.pin2, sda=microbit.pin1)` to switch the pins, the first two calls to `microbit.i2c.write()` always fail with `OSError: I2C write failed with error code -1010` (and no data is sent on the pins), but afterwards it all works fine. Right now I'm just calling an empty write twice and catch the exception to work around that, but it might be a symptom of some state not being cleaned somewhere.

deshipu, 2017-10-15:
Because I need the `i2c.init()`, I'm using the current master version.

carlosperate, 2017-10-15:
If you could try `version1` branch it'd be quite useful, as that's where all the new development is happening. The i2c init logic is still very similar, so it's very likely the problem would still be present there.

I wonder if the peripheral needs a little bit of time to stabilise, could you try adding a delay between the init and write? 

As a side note, in case this function needs updating, I was looking at the init code and I'm not sure if the comment in the frequency() line is still valid:
https://github.com/bbcmicrobit/micropython/blob/e88e3b2f6c579eab0eafda02242092bff51f6934/source/microbit/microbiti2c.cpp#L78

Looking at the mbed/nordic code, I don't see any reset there. From mbed:
```c
void I2C::frequency(int hz) {
    _hz = hz;

    // We want to update the frequency even if we are already the bus owners
    i2c_frequency(&_i2c, _hz);

    // Updating the frequency of the bus we become the owners of it
    _owner = this;
}
```
https://github.com/NordicSemiconductor/mbed/blob/master/libraries/mbed/common/I2C.cpp#L32-L40

And nRF51 target:
```c
void i2c_frequency(i2c_t *obj, int hz)
{
    if (hz<250000) {
        obj->freq           = 100000;
        obj->i2c->FREQUENCY = (TWI_FREQUENCY_FREQUENCY_K100 << TWI_FREQUENCY_FREQUENCY_Pos);
    } else if (hz<400000) {
        obj->freq           = 250000;
        obj->i2c->FREQUENCY = (TWI_FREQUENCY_FREQUENCY_K250 << TWI_FREQUENCY_FREQUENCY_Pos);
    } else {
        obj->freq           = 400000;
        obj->i2c->FREQUENCY = (TWI_FREQUENCY_FREQUENCY_K400 << TWI_FREQUENCY_FREQUENCY_Pos);
    }
}
```
https://github.com/NordicSemiconductor/mbed/blob/master/libraries/mbed/targets/hal/TARGET_NORDIC/TARGET_NRF51822/i2c_api.c#L149-L162

deshipu, 2017-10-15:
I can reproduce the same problem with the `version1` branch. A delay doesn't seem to help, though it seems that then only one transaction fails, not two — but the first transaction fails always. So maybe there is both some state problem *and* settling down.

I just tried another thing: change the frequency, but don't change the pins (leave it on the internal I²C bus). In this case the first write to a random address succeeds, even when there is no such device. The second write fails as it should...

deshipu, 2017-10-15:
I added a call to `i2c_reset()` in the `master` branch (because there I can access the _i2c object, in a hacky way), and that seems to solve the issue. Not sure how to port that to the `version1` branch, though, as the dal doesn't expose this.

dpgeorge, 2017-10-27:
Thanks @deshipu for the report.  I couldn't reproduce exactly the issues your reported, but it did find similar ones.  In particular the following code shows the bug:
```python
i2c.init(scl=pin0, sda=pin1) # doesn't actually change the pins!
i2c.scan() # this will call i2c_reset() which will change the pins
i2c.init() # doesn't actually reset the pins to the original internal ones!
i2c.read(14, 1) # this should read the internal I2C chip but gives an error!
```
As @carlosperate points out (thanks!), the i2c_frequency() function doesn't reset the i2c device.  And in fact changing the pins in i2c.init() didn't really change them properly, it just changed the values in the i2c_t struct.  So the new pins were only becoming active after an error, because that's when i2c_reset() is called.

The solution was indeed to call i2c_reset() in the i2c.init() method, and this is done on the version1 branch in 1ef12d220e597e8a4c62a814d78b2a819835f7ad

closed 2017-10-27 by dpgeorge