bbcmicrobit/micropython #475: Can't access REPL when an error is scrolling.

deshipu, 2017-10-15:
I tested it on the `master` and `version1` branches. When the flashed Python program raises an exception, it's scrolled on the 5×5 display of the micro:bit. The problem is that until it stops scrolling (and that can take a bit of time), it's not possible to access the REPL console. Neither `ctrl+c` nor `ctrl+d` work, the only way to debug the issue that caused the exception is to patiently wait for the scrolling to finish, which is pretty frustrating.

dpgeorge, 2017-10-15:
I can confirm this issue and agree it would be good to allow to break out of the scrolling error.  But @jaustin had some other ideas about how to improve these error messages: @jaustin what are we doing about error messages in version1?  If we are not changing them then it would be good to at least fix the issue here, to allow ctrl-C to stop a scolling message.

jadudm, 2017-11-08:
Just to add, tools like 'ufs ls' also timeout/error out when this is happening. It is particularly problematic for students/users of Mu, where Mu simply crashes when you try and (for example) show the files on the device while your code is in an error state.

I mention this only because there is no real CTRL-C option in Mu; a student can interact with the GUI at any point, and trying to show files while an error condition exists seems to reliably cause an error.

(This may be more a Mu issue than MicroPython problem, but remembering other modalities may help inform the decision on a fix/patch/etc.)

DavidWhaleMEF, 2018-12-13:
@dpgeorge Does the v1.0.0 micropython fix this issue? And if so, can we close this ticket now?

jadudm, 2018-12-13:
I was a follow-on to this ticket. I have not touched this for some time, but I will set things up and give it a test to confirm whether there is still a problem, if that helps your process.

I can test tools on a Mac, but do not have a native Windows (or Linux) machine to test from. (I can test from a virtual machine for Win/Linux; I don't know how that impacts subsystems like USB for purposes of this test. It would probably be OK.)

microbit-carlos, 2018-12-14:
Ctrl+C to stop an error message has been implemented in https://github.com/bbcmicrobit/micropython/commit/13426012b5a314367c9592ae5b3d70ca4023a524 (included since the first v1.0.0-beta.1).

Tools like Mu and uFs will try to access the REPL and send Ctrl+C commands, so if using a MicroPython version with that commit included they should connect right away even if an exception is scrolling.

The current version of Mu also sends a user-triggered Ctrl+C command via the REPL panel, so it should work there as well.

Since this has been done already I will close this issue, but please feel free to reopen or add more comments it if there is a problem or you have any more questions.

closed 2018-12-14 by microbit-carlos