bbcmicrobit/micropython #477: Need more detail in uart docs

DavidWhaleMEF, 2017-12-10:
Relevant to this issue: https://github.com/bbcmicrobit/micropython/issues/472

The UART docs need more details, such as:
- what is the receive timeout duration
- can you change the receive timeout duration?
- the fact that sending an 0x03 via USBSER will KeyboardInterrupt your program
- whether the CTRL-C processing is enabled or disabled when the UART is redirected to the pins.
- how/if you can change the CTRL-C processing to enable binary data transfer in all of the above (and if it is not possible to change, state the fact that it cannot be changed)

DavidWhaleMEF, 2018-01-30:
@dpgeorge I've been trying to locate the default timeout duration of uart.readline() in the source, but got a bit wrapped up in all the abstraction levels with streams and the HAL and can't find it. Can you confirm what the timeout period is for readline() please? Thanks!

dpgeorge, 2018-02-07:
@DavidWhaleMEF the timeout for all UART reads is a value that depends on the baudrate and is otherwise not changeable via Python.  It is: `microbit_uart_timeout_char = 13000 / baudrate + 1`.  It is done on a per-character basis, so that's the longest it will wait for a new character to arrive.  And that should be the timeout for readline() if a newline doesn't appear.

microbit-rosslowe, 2018-10-18:
1. Add a note to uart.rst to clarify receive timeout duration (see above)
2. Note "that sending an 0x03 via USBSER will KeyboardInterrupt your program"
3. Add reference to this:

[image]

@microbit-carlos I can make a branch for these, I'm not sure on "whether the CTRL-C processing is enabled or disabled when the UART is redirected to the pins."

microbit-carlos, 2018-10-18:
The CTRL-C is enabled disabed via `kbd_int`, so adding a reference to that method in the uart docs would be enought.

PR #586 (2018-10-19): docs: Update uart.rst with additional detail
- add note on receive timeout duration
- add note on 0x03 keyboard interrupt
- add reference to micropython.kbd_intr() in relation to keyboard interrupts

Closes #477
closed 2018-10-19 by microbit-carlos