bbcmicrobit/micropython #480: ValueErrors when writing analog to pins

MaximusWuf, 2018-01-02:
Microbit report value errors when writing analog to pins, saying the value should be between 0 and 1023.
The input is an integer number:
Example code:
`voltage = 512
while True:
    if button_a.is_pressed():
        pin14.write_analog(voltage)
        pin15.write_analog(voltage)
    if button_b.is_pressed():
        voltage = voltage + 100`
When button a is solely pressed, it works fine, however if b were to be pressed and then a afterwards, it would report a value error. (Press b before pressing a and it would still fail)

`i = 100
while True:
    if button_a.is_pressed():
        i += 100
        pin14.write_analog(i)`

This would also fail.

DavidWhaleMEF, 2018-01-03:
Yes, this is correct behaviour for the above code.

After a few milliseconds of holding button A in the second example, you would have added several chunks of 100 to i and it will now be bigger than 1023, leading to the value error. Your while True loop will spin round very quickly.

Either use a sleep(100) to slow things down a bit after the pin14.write_analog(i), or use button_a.was_pressed() to sense a single press/release transition, or add this code to cap it's value to prevent overflow:

```
if i+100 < 1023:
    i += 100
```

Look also at the docs to understand the difference between button_a.is_pressed() and button_a.was_pressed() - I suspect you want was_pressed() in this case.

dpgeorge, 2018-03-07:
I assume the answer above helped resolve the issue.

closed 2018-03-07 by dpgeorge