bbcmicrobit/micropython #481: Temperature() locks up in v1.9.2

mcauser, 2018-01-05:
```
MicroPython v1.9.2-34-gd64154c73 on 2017-09-01; micro:bit with nRF51822
Type "help()" for more information.
>>> from microbit import *
>>> temperature()
```
Unresponsive, can't CTRL+C or CTRL+D. Need to press HW reset button.

microbit-carlos, 2018-01-05:
I can confirm the same behaviour on micro:bit v1.0-[e17de09](https://github.com/bbcmicrobit/micropython/commit/e17de0954178d88aa2a1d70b6e6ebc43f5456607) (latest commit for v1 ) from http://python.microbit.org/v/beta

dpgeorge, 2018-01-31:
Thanks for the report, I can confirm that it's an issue.  It seems to be because the underlying DAL implementation is calling into the soft device, which doesn't exist in the uPy build.  Should now be fixed in the version1 branch by commit c5e84ec37294f125f18354e669effce097871e91

Leaving this issue open until version1 is merged back into master.

dpgeorge, 2018-03-07:
Fixed in merge of version1 branch at e26d7c89d4a96de0fa0a1dd5aec024b31fc4816e

closed 2018-03-07 by dpgeorge