bbcmicrobit/micropython #483: MicroBitImage.blit - no attribute

DavidWhaleMEF, 2018-01-12:
mu: 0.9.13

The docs state that MicroBitImage has a blit() operation, but the installed and public version of MicroPython does not support this operation.

```
>>> TRACE = Image('00000:00000:00000:90000:99990:00090:00090:00090:00090:00090:')
>>> i = Image()
>>> i.blit(TRACE, 0,0,5,5,0,0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'MicroBitImage' object has no attribute 'blit'
>>> dir(i)
['width', 'height', 'get_pixel', 'set_pixel', 'shift_left', 'shift_right', 'shift_up', 'shift_down', 'copy', 'crop', 'invert', 'HEART', 'HEART_SMALL', 'HAPPY', 'SMILE', 'SAD', 'CONFUSED', 'ANGRY', 'ASLEEP', 'SURPRISED', 'SILLY', 'FABULOUS', 'MEH', 'YES', 'NO', 'CLOCK12', 'CLOCK1', 'CLOCK2', 'CLOCK3', 'CLOCK4', 'CLOCK5', 'CLOCK6', 'CLOCK7', 'CLOCK8', 'CLOCK9', 'CLOCK10', 'CLOCK11', 'ARROW_N', 'ARROW_NE', 'ARROW_E', 'ARROW_SE', 'ARROW_S', 'ARROW_SW', 'ARROW_W', 'ARROW_NW', 'TRIANGLE', 'TRIANGLE_LEFT', 'CHESSBOARD', 'DIAMOND', 'DIAMOND_SMALL', 'SQUARE', 'SQUARE_SMALL', 'RABBIT', 'COW', 'MUSIC_CROTCHET', 'MUSIC_QUAVER', 'MUSIC_QUAVERS', 'PITCHFORK', 'XMAS', 'PACMAN', 'TARGET', 'ALL_CLOCKS', 'ALL_ARROWS', 'TSHIRT', 'ROLLERSKATE', 'DUCK', 'HOUSE', 'TORTOISE', 'BUTTERFLY', 'STICKFIGURE', 'GHOST', 'SWORD', 'GIRAFFE', 'SKULL', 'UMBRELLA', 'SNAKE']
>>> 
```

DavidWhaleMEF, 2018-01-12:
@carlosperate I recall that you moved the label on the published docs recently to make the docs match the current public release of MicroPython - can you double check that the label is correct and that the docs for the matching public release are in step please, as this problem above indicates that docs and code are still out of step. Thanks.

cc @jaustin

DavidWhaleMEF, 2018-01-12:
Here is what the docs currently say. This does not match with the reality of what the current public release of MicroPython offers.

[image]

ntoll, 2018-01-12:
Looking at the current docs on RTD... they're completely out of date, missing a large part of the tutorial I wrote and likely completely inconsistent with other aspects of MicroPython.

Someone (who) changed the, to a *really old* 0.9 version. I've updated the settings so they're now pointing at latest.

markshannon, 2018-01-12:
The problem isn't the docs, but that the version of MicroPython is really out of date.
The blit method was committed a year ago in bfbfe87b

DavidWhaleMEF, 2018-01-12:
@ntoll @carlosperate @markshannon Whatever the reason, we need the docs to always be in step with the version of code deployed at python.microbit.org and codewith.mu (current 0.9.13 for Mu)
No idea for python.microbit.org (the URL says /v/1 but I have no idea what version of code or docs that relates to - how do we check?)

markshannon, 2018-01-12:
> Whatever the reason, we need the docs to always be in step with the version of code deployed at python.microbit.org

Absolutely, yes.
But isn't that a microbit.org issue? They should deploy the docs and executable built from the same checkout.

DavidWhaleMEF, 2018-01-12:
For some reason @carlosperate tried to move the tag, but I don't think he had access to do that. @ntoll can you make sure that @carlosperate has the necessary access rights please? Thanks.

ntoll, 2018-01-12:
@DavidWhaleMEF access rights to what? You either mean to the Git repos or the ReadTheDocs setup... hard to tell which you mean. ;-) (Happy to do so once I know.)

ntoll, 2018-01-12:
FYI... I'm no longer have admin rights on this repos -- you'll have to ask @dpgeorge.

jaustin, 2018-01-12:
I can also add @microbit-carlos, @dpgeorge it's your project though so gimmie and Ack and I'll do the admin :)

deshipu, 2018-01-13:
> Someone (who?) changed the, to a really old 0.9 version. I've updated the settings so they're now pointing at latest.

I did that, because I was asked to do it, and because it made sense. It
doesn't matter what awesome new features Micro:bit MicroPython has — as
long as they are not released they are useless for the average users.
The people who can compile and flash the newest version from the
repository can also find the version selector on Readthedocs, I think.
For all the rest, defaulting to what they are actually using makes the
most sense.

We had several posts in here and on the list already asking about things that are in the docs but
don't actually exist in the released versions. Making the released version of
the docs the default fixes that.

If you have some version-independent changes in the documentation that you
would like to make available anyways, I think it would be best to backport it.

ntoll, 2018-01-13:
@deshipu aha... I'd totally missed that. I wasn't trying to point fingers...! ;-)

Ultimately, we need the work from @dpgeorge to be released so *everything* (docs and code) is up to date -- there's a lot of unreleased work that's happened over the past 18 months, it's a shame it's not being used. I think we're waiting on @jaustin to give guidance to Damien for what things need fixing.

microbit-carlos, 2018-03-08:
Created https://github.com/bbcmicrobit/micropython/issues/498 to make sure we have the correct default docs until v1.0 is released.

closed 2018-03-08 by microbit-carlos