bbcmicrobit/micropython #484: is_touched(). Is it capacitance or resistance?

ukBaz, 2018-01-13:
The description for is_touched() in [pin.rst](https://github.com/bbcmicrobit/micropython/blob/master/docs/pin.rst#175) says the following:

> This test is done by measuring the capacitance of the pin together with
        whatever is connected to it. Human body has quite a large capacitance,
        so touching the pin gives a dramatic change in reading, which can be
        detected.

While the runtime documentation at https://lancaster-university.github.io/microbit-docs/ubit/io/#overview says that it is *basic touch sensing through resistance detection*

I suspect it is resistance. Does anyone know?

DavidWhaleMEF, 2018-01-13:
It is resistive touch sense. The docs are wrong.

That's why if you look at the schematic you will see very weak pullup resistors on P0 P1 P2, as documented here: http://tech.microbit.org/hardware/edgeconnector_ds/

DavidWhaleMEF, 2018-01-30:
Has anyone updated the docs on this yet do you know @ukBaz ? If so, we can close this issue down?

ukBaz, 2018-01-30:
The docs have not been updated.

dpgeorge, 2018-01-31:
Thanks for the report.  I fixed the docs in c958e03f5c9255a771a3d8ddb68b9969ec276871

closed 2018-01-31 by dpgeorge