bbcmicrobit/micropython #486: Neopixel methods working incorrectly in MicroPython v1.9.2-34

jafine, 2018-01-14:
There seems to be a neopixel issue with the above version of MicroPython, that is bundled with the mu beta.

* This should be able to be reproduced with MicroPython v1.9.2-34-gd64154c73
* Previously-working neopixel code that runs .clear() or .show(), results in all neopixels appearing to be set to (255, 255 ,255), rather than the set values.
* Reverting to an older version of MicroPython, i.e. 3.4.2 from the python web interface, using the same code, works as expected.

Example code that typically triggers this would be:
```python
 import neopixel

 np = neopixel.NeoPixel(pin0, 5)
 np[0] = (255,0,0)  # should make the first LED red
 np.show()  # but this makes all LEDs go white, i.e. (255,255,255)
```
calling `.clear()` gives the same result.

This was observed on Kitronik ZIP LEDs in the halo and servo lite boards, and although I assume these are essentially connecting straight through to the LEDs, it may be prudent to confirm with RGB LEDs directly connected to a micro:bit.

microbit-carlos, 2018-03-08:
@microbit-mark could you try to replicate this one?

microbit-mark, 2018-03-20:
I can't replicate. I get normal behaviour using a  micro:pixel edge connector using MicroPython v1.9.2-34-gd64154c73

AlasdairAtKitronik, 2018-04-18:
This same issue has been observed repeatedly on WS2812B-MINI LEDs.
Looking into the datasheets for both the WS2812B and WS2812B-MINI LEDs, it can be noted that the Data Transfer Times for 0s and 1s is different for the two LED size variants.

WS2812B: 0 data line high time = 250 - 550 ns	1 data line high time = 700 - 1000 ns (https://www.kitronik.co.uk/pdf/WS2812B-LED-datasheet.pdf)
WS2812B-MINI: 0 data line high time = 220 - 380 ns	1 data line high time = 580 - 1600 ns (https://www.tme.eu/gb/Document/7f6560690095a6f285728f6a415c4f88/WS2812B-MINI.pdf)

Running MicroPython code to set all the LEDs to OFF (i.e. (0, 0, 0)) leads to all LEDs being ON White Full Brightness (i.e. (255, 255, 255)).

We believe this is due to the MicroPython NeoPixel code setting the data line high time for 0s to longer than the WS2812B-MINI LED allowable range for a 0 data line high time (approx 560ns).
With the addition of a few ns extra delay, potentially caused by other components, the time the data line is held high shifts to the range set for a 1, and therefore the LEDs turn on white.

DaveAtKitronik, 2018-04-26:
Just compiling neopixel.c with -O3 for speed optimisation and a bit of loop unrolling improves the timings by enough that both LEDs will work. It adds *LOADS* of size, so isn't a workable fix, but proves it is worth some effort as it can work.

DaveAtKitronik, 2018-04-26:
The only function that needs optimising is neopixel_show (...) which is responsible for clocking out the bits. neopixel_clear sets the data for the LEDs to 0 and then calls neopixel_show.

As currently written neopixel_show uses a series of NOPs to create pauses between switching the line state from high to low. 
WS2812 use a NRZ protocol, where a '0' is a short high and a long low, and a '1' is a long high and a short low. 
As described by @AlasdairAtKitronik above there is overlap between the 'B' and the 'mini' versions - the sweet spot being around 300nS and 800nS respectively to cover both chips and be nicely in tolerance. 
Makecode uses assembler to create pulses that are ~320nS for the short ( '0' ) pulse. 
Longer pulses ( '1') are created by leaving the line high whilst preparing for the next bit and then taking it low. 
By transplanting the makecode assembler (with appropriate changes) into the neopixel_show routine we have managed to take the following measurements: 
'0' high time of ~320ns, and '1' high time of ~800ns. 
Both of these are within spec for both types of neopixel, and not very close to either edges, such that subsequent changes caused by component tolerance or compiler optimisations are unlikely to push them out of spec.
The first WS2812Mini in the chain regenerates these to pulses of 360nS and 680nS respectively.
Brief testing using both a Kitronik :GAME ZIP64 (WS2812Mini) and a Kitronik :MOVE Servo:Lite (WS2812B) shows that the test case functions as expected. The test case turns on the first 4 LEDs with different colours chosen to give distinctive memory patterns in the debugger.
Testing using a :GAME ZIP64 with a level switch FET that is slower than the correct fitted one shows the '0' pulse to be stretched to ~400nS
This is outside specification, but the LED's still function correctly as it is still quite a lot smaller than the 580nS minimum for a '1' 
Minor changes were needed to the makecode approach to cope with different data types and such. 
I will fork this repo tomorrow and commit the changes for a pull request. 
Tagging @dpgeorge @microbit-carlos for reference

dpgeorge, 2018-05-04:
This should now be fixed by 4fc103da6e612569de53dbdb61edf843a3677320

closed 2018-05-04 by dpgeorge

microbit-carlos, 2018-05-04:
The fixes can be tested at http://python.microbit.org/v/beta

Thanks again @DaveAtKitronik @AlasdairAtKitronik and @dpgeorge!

DaveAtKitronik, 2018-05-08:
Just tested http://python.microbit.org/v/beta on a :Game Zip64 - all good :) 
Cheers for a speedy turn around @dpgeorge and @carlosperate

dpgeorge, 2018-05-08:
Thanks @DaveAtKitronik for testing.