bbcmicrobit/micropython #487: Documentation for I/O Pin classes is incorrect for digital PWM  support

MrYsLab, 2018-01-17:
write_analog, set_analog_period and set_analog_period_microseconds should be moved from [class microbit.MicroBitAnalogDigitalPin](http://microbit-micropython.readthedocs.io/en/latest/pin.html#microbit.MicroBitAnalogDigitalPin) to [class microbit.MicroBitDigitalPin](http://microbit-micropython.readthedocs.io/en/latest/pin.html#microbit.MicroBitDigitalPin) so that the hierarchy flows correctly.

In addition, set_pull should be added to class microbit.MicroBitDigitalPin

deshipu, 2018-01-17:
Would you be willing to make those changes and submit a pull request? We can help you with that.

MrYsLab, 2018-01-17:
No problem. I will submit ASAP (probably within a couple of days). I am familiar with pull requests but will ask for help if I run into any issues.

microbit-carlos, 2018-10-09:
@MrYsLab We are working through the documentation at the moment, are you still interested to create a PR for this?

MrYsLab, 2018-10-09:
Yes, PR #488 was submitted to address the issue.

MrYsLab, 2018-10-10:
@microbit-carlos Going over this, it may not be clear as to my rationale. When I created [pseudo-microbit](https://github.com/MrYsLab/pseudo-microbit]) I tried following the documentation to create the pin class so that PyCharm could do auto-completion. Following the documentation as it is currently written, when I created the classes, things did not work. I had to modify the pin class definitions to match my request. There is also some clean-up reorganization (IMHO) to make things easier to understand.

microbit-carlos, 2018-10-11:
Thanks for the clarification @MrYsLab and I agree, since all pins do contain the write_analog group of methods they should be part of the MicroBitDigitalPin section.
Thanks for sending the PR!

microbit-carlos, 2018-10-19:
PR https://github.com/bbcmicrobit/micropython/pull/488 has been merged, thanks for all the effort @MrYsLab!

closed 2018-10-19 by microbit-carlos