bbcmicrobit/micropython #489: How to use the LightLevel in micropython?

ouki-wang, 2018-01-19:
In https://makecode.microbit.org/ ,I find that we can use input.lightLevel to get the date. How could I use this function by micropython.Thanks a lot!

DavidWhaleMEF, 2018-01-19:
There is no equivalent feature in MicroPython currently.

@markshannon and @zeth did a prototype at a previous PyCon sprint, but this has not surfaced into live code yet. Sorry.

DavidWhaleMEF, 2018-01-30:
The DAL does this by turning a number of LEDs output-high, then to an analog input and measuring the analog voltage at a precise time in the future. The collapse time of the voltage on the LED is roughly proportional to a red-sensitive version of ambient light. The timing is done via a scheduled timer interrupt, which is hard to do in Python unless you rebuild the C code,

Timing how long before the pin falls below the on/off threshold is another way to do this if precise timing is not available, although the decay time is quite fast and the python interpreter loop probably won't get round fast enough. You could stuff some assembler inside a function that sits in a tight loop counting how many times it goes round the loop until the voltage collapses below the 1/0 pin threshold, and the assembler for that would be relatively small.

DavidWhaleMEF, 2018-01-30:
A much simpler (and cheap) method is to attach an LDR to one of the analog pins. Although it is then not a built-in solution.

DavidWhaleMEF, 2018-01-30:
I'm not sure we can offer much more help on this issue at this time. Can we close it perhaps?

dpgeorge, 2018-01-31:
This is classified as an enhancement, and is actually a duplicate of #466.  So let's close this and continue any discussion in that prior ticket.

closed 2018-01-31 by dpgeorge