bbcmicrobit/micropython #492: func.__name__ not supported

MrYsLab, 2018-02-03:
This was corrected in MicroPython https://github.com/micropython/micropython/issues/614
but apparently, the fix was not brought over to the micro:bit.

dpgeorge, 2018-02-07:
This feature does exist in the micro:bit code but it's not enabled.  To enable it needs the option MICROPY_PY_FUNCTION_ATTRS.  It's not enabled because it takes up precious space and is not seen to be something that is necessary on the micro:bit.

What use case do you have for `__name__`?

MrYsLab, 2018-02-07:
I wanted to write a decorator to do tracing that would write to persistent storage for debugging. Is MICROPY_PY_FUNCTION_ATTRS something I can set in my script or is it a flag used in building MicroPython? If I can turn it on/off at will that would suit my purposes.

dpgeorge, 2018-02-07:
> Is MICROPY_PY_FUNCTION_ATTRS something I can set in my script or is it a flag used in building MicroPython?

It's a compile-time setting, so you'd need to rebuild the firmware (.hex) file if you want to enable it.  To enable it edit the file inc/microbit/mpconfigport.h to add `#define MICROPY_PY_FUNCTION_ATTRS (1)`

MrYsLab, 2018-02-07:
Thanks for the answer. I will close this request due to the space constraints.

closed 2018-02-07 by MrYsLab